riscv-collab/riscv-gnu-toolchain #860: Compilation failed. g++: error: spec failure: unrecognized spec option ‘@’

zinechant, 2021-03-09:
Ubuntu 18.04, gcc 7.5.0, configured with `./configure --prefix=$HOME/.riscv`

Compilation error:
```
if /home/zinechant/riscv-gnu-toolchain/build-gcc-newlib-stage1/./gcc/xgcc -B/home/zinechant/riscv-gnu-toolchain/build-gcc-newlib-stage1/./gcc/ -print-sysroot-headers-suffix > /dev/null 2>&1; then \
  set -e; for ml in `/home/zinechant/riscv-gnu-toolchain/build-gcc-newlib-stage1/./gcc/xgcc -B/home/zinechant/riscv-gnu-toolchain/build-gcc-newlib-stage1/./gcc/ -print-multi-lib`; do \
    multi_dir=`echo ${ml} | sed -e 's/;.*$//'`; \
    flags=`echo ${ml} | sed -e 's/^[^;]*;//' -e 's/@/ -/g'`; \
    sfx=`/home/zinechant/riscv-gnu-toolchain/build-gcc-newlib-stage1/./gcc/xgcc -B/home/zinechant/riscv-gnu-toolchain/build-gcc-newlib-stage1/./gcc/ ${flags} -print-sysroot-headers-suffix`; \
    if [ "${multi_dir}" = "." ]; \
      then multi_dir=""; \
    else \
      multi_dir=/${multi_dir}; \
    fi; \
    echo "${sfx};${multi_dir}"; \
  done; \
else \
  echo ";"; \
fi > tmp-fixinc_list
mv tmp-libgcc.mvars libgcc.mvars
/bin/bash ../.././riscv-gcc/gcc/../move-if-change tmp-fixinc_list fixinc_list
echo timestamp > s-fixinc_list
g++ -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -Ilto -I../.././riscv-gcc/gcc -I../.././riscv-gcc/gcc/lto -I../.././riscv-gcc/gcc/../include -I../.././riscv-gcc/gcc/../libcpp/include  -I../.././riscv-gcc/gcc/../libdecnumber -I../.././riscv-gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././riscv-gcc/gcc/../libbacktrace   -o lto/lto-dump.o -MT lto/lto-dump.o -MMD -MP -MF lto/.deps/lto-dump.TPo ../.././riscv-gcc/gcc/lto/lto-dump.c
g++: error: spec failure: unrecognized spec option ‘@’
cc1plus: fatal error: {I*&F*}: No such file or directory
compilation terminated.
Makefile:1117: recipe for target 'lto/lto-dump.o' failed
make[2]: *** [lto/lto-dump.o] Error 1
```

Didn't find much about g++ spec option via Google

jim-wilson, 2021-03-09:
This is an odd error.

Internally, GCC uses a specs file to control how command line options get passed from the driver (gcc, g++, etc) to the subprograms (cc1, as, ld, etc).  Normally, this specs file is inside the driver binary, but you can optionally create a specs file for the driver to load at run-time.  You apparently have a specs file for a gcc version greater than 7.5 someplace where the system gcc-7.5 can find, and it is complaining about the syntax since new gcc versions use spec syntax older gcc versions don't support.  You need to figure out where this specs file is, and take steps so that the system gcc won't use it.

You can add a -v option to the g++ command that is failing, and in the verbose output it will tell you that it is using a specs file, and where it found it.  Then depending on that result you need to make changes to avoid the problem.  Maybe you have some environment variables set that are confusing gcc for instance.

zinechant, 2021-03-09:
Thanks for the reply.

This time I cleared CPATH, LD_LIBRARY_PATH, LIBRARY_PATH environment variables. But still got the same error.

Seems the spec file involved is inside the repo. No external spec file.

```
g++ -v -fno-PIE -c   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -I. -Ilto -I../../../riscv-gcc/gcc -I../../../riscv-gcc/gcc/lto -I../../../riscv-gcc/gcc/../include -I../../../riscv-gcc/gcc/../libcpp/include  -I../../../riscv-gcc/gcc/../libdecnumber -I../../../riscv-gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../../../riscv-gcc/gcc/../libbacktrace   -o lto/lto-dump.o -MT lto/lto-dump.o -MMD -MP -MF lto/.deps/lto-dump.TPo ../../../riscv-gcc/gcc/lto/lto-dump.c
Reading specs from ./specs
COLLECT_GCC=g++
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.5.0-3ubuntu1~18.04' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc driver version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04) executing gcc version 10.2.0
g++: error: spec failure: unrecognized spec option ‘@’
COLLECT_GCC_OPTIONS='-v' '-fno-PIE' '-c' '-g' '-O2' '-D' 'IN_GCC' '-D' 'CROSS_DIRECTORY_STRUCTURE' '-fno-exceptions' '-fno-rtti' '-fasynchronous-unwind-tables' '-Wextra' '-Wall' '-Wno-narrowing' '-Wwrite-strings' '-Wcast-qual' '-Wsuggest-attribute=format' '-Woverloaded-virtual' '-Wpedantic' '-Wno-long-long' '-Wno-variadic-macros' '-Wno-overlength-strings' '-D' 'HAVE_CONFIG_H' '-I' '.' '-I' 'lto' '-I' '../../../riscv-gcc/gcc' '-I' '../../../riscv-gcc/gcc/lto' '-I' '../../../riscv-gcc/gcc/../include' '-I' '../../../riscv-gcc/gcc/../libcpp/include' '-I' '../../../riscv-gcc/gcc/../libdecnumber' '-I' '../../../riscv-gcc/gcc/../libdecnumber/dpd' '-I' '../libdecnumber' '-I' '../../../riscv-gcc/gcc/../libbacktrace' '-o' 'lto/lto-dump.o' '-MT' 'lto/lto-dump.o' '-MMD' '-MP' '-MF' 'lto/.deps/lto-dump.TPo' '-shared-libgcc' '-mtune=generic' '-march=x86-64'
 /usr/lib/gcc/x86_64-linux-gnu/7/cc1plus -quiet -v {I*&F*} -MMD lto/lto-dump.d -MF lto/.deps/lto-dump.TPo -MP -MT lto/lto-dump.o -D_GNU_SOURCE -D IN_GCC -D CROSS_DIRECTORY_STRUCTURE -D HAVE_CONFIG_H ../../../riscv-gcc/gcc/lto/lto-dump.c -quiet -dumpbase lto-dump.c -mtune=generic -march=x86-64 -auxbase-strip lto/lto-dump.o -g -O2 -Wextra -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wsuggest-attribute=format -Woverloaded-virtual -Wpedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings -version -fno-PIE -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -fstack-protector-strong -Wformat-security -o /tmp/cc2TKZ7s.s
GNU C++14 (Ubuntu 7.5.0-3ubuntu1~18.04) version 7.5.0 (x86_64-linux-gnu)
        compiled by GNU C version 7.5.0, GMP version 6.1.2, MPFR version 4.0.1, MPC version 1.1.0, isl version isl-0.19-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
ignoring nonexistent directory "/usr/lib/gcc/x86_64-linux-gnu/7/../../../../x86_64-linux-gnu/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/include/c++/7
 /usr/include/x86_64-linux-gnu/c++/7
 /usr/include/c++/7/backward
 /usr/lib/gcc/x86_64-linux-gnu/7/include
 /usr/local/include
 /usr/lib/gcc/x86_64-linux-gnu/7/include-fixed
 /usr/include
End of search list.
cc1plus: fatal error: {I*&F*}: No such file or directory
compilation terminated.
```

And another error also surface:
```
 g++ -v -no-pie   -g -O2 -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -static-libstdc++ -static-libgcc  -o lto1 lto/lto-lang.o lto/lto.o lto/lto-object.o attribs.o lto/lto-partition.o lto/lto-symtab.o lto/lto-common.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a  -lmpc -lmpfr -lgmp -rdynamic -ldl  -lz  libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a
Reading specs from ./specs
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
g++: error: unrecognized command line option ‘-rdynamic’
[... cut ...]  --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc driver version 7.5.0 (Ubuntu 7.5.0-3ubuntu1~18.04) executing gcc version 10.2.0
```

The last line caught my eyes. Don't think there is a gcc 10.2.0 on my system. Does this repo do some kind of bootstrapping which gives birth to this gcc 10.2.0

jim-wilson, 2021-03-09:
The compiler version comes from the specs file.  Every problem here is related to the bad specs file.  g++ is saying "./specs" but the system compiler should not be using that file.  This is the specs file for the riscv cross compiler you are building.  This does not appear to be a problem with PATH.  I think there is still some environment variable problem.  G++ is reporting that OFFLOAD_TARGET_NAMES=nvptx-none, so maybe you set environment variables for the nvptx cross compiler that are confusing the system compiler.

You can see the search path for a specs file by using "g++ --print-search-dirs" and then looking at the libraries: line.  The dir where you are building gcc should not be on that path.  This dir has build-gcc-* in its name so this is pretty unlikely, but it is possible that "." might be on the path which would definitely cause trouble.

Some env vars that might cause trouble are GCC_EXEC_PREFIX, LIBRARY_PATH_ENV, and LPATH,.  There might also be others I've forgotten.

Another way to get this error would be to pass -B./ to gcc, but I don't see that on the command line.

Generally, it is a bad idea to set enviroment variables that change gcc behavior, as you can forget that you set it, and then you get strange errors that no one else can reproduce.

zinechant, 2021-03-09:
Thanks.

The problem seems relate to that I set `LIBRARY_PATH` in `.bashrc` to include `$HOME/.local/lib`.
Somehow gcc would pickup this setting and generate a path starts with . into search path.

The issue is resolved now. I will close it.

closed 2021-03-09 by zinechant