riscv-collab/riscv-gnu-toolchain #861: Issue with installation for linux

RaoShri, 2021-03-10:
Hello,
I am trying to install riscv6-toolchain for linux, I am getting the following error.

$ echo $RISCV
/opt/riscv/bin

$ ./configure --prefix=$RISCV   
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for fgrep... /usr/bin/grep -F
checking for grep that handles long lines and -e... (cached) /usr/bin/grep
checking for bash... /bin/bash
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... /usr/bin/ftp
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed

$ sudo make linux
make: Nothing to be done for 'linux'.

I am new at this. Please help. Also could you tell me the difference between newlib and Linux installation? How to decide which one I need? I am right now simulating Rocketchip on Chipyard on a Linux platform

Regards,
Shrinidhi

pz9115, 2021-03-10:
Hi RaoShri,

It's not an error when you want to use 'make' the toolchain again. If you want to make again, you need use 'make clean' before next 'make'. 

The newlib and glibc are both C library, Here is the introduce of them:

https://sourceware.org/glibc/wiki/HomePage

https://sourceware.org/newlib/README

You can use any of them on your platform, and the newlib will generator more small codesize.

jim-wilson, 2021-03-11:
There are some feature differences between the embedded elf and linux toolchains.  The linux toolchain uses glibc, the embedded elf one uses newlib.  The linux toolchain supports shared libraries, the embedded elf one doesn't.  The linux toolchain has a full C library, the embedded elf one supports only the most common C library routines.

Generally, if you are compiling linux software, you must use a linux toolchain.  If you want to keep things simple, then use an embedded elf toolchain.

closed 2022-09-27 by TommyMurphyTM1234