riscv-collab/riscv-gnu-toolchain #862: Inline Assembly - Already added new instruction to binutils

Nado15, 2021-03-11:
Hello, 

I have added a new instruction to binutils but I would like to call this instruction via inline assembly. I tried but gcc is throwing following error if I compile via `riscv64-unknown-elf-gcc main.c`
`main.c: Assembler messages:
 main.c:4: Error: unrecognized opcode *addone x2,0* `

But if I compile like 
`riscv64-unknown-elf-gcc main.c -S -o main.s`

and add an extra step

`riscv64-unknown-elf-as main.s main`

It does work. 

What do I have to do that gcc is accepting my new instruction "addone" via inline assembly in C-code directly without the extra step?

jim-wilson, 2021-03-11:
If you add the -v option when compiling, gcc will tell you which assembler it is using.  Perhaps it is using a different assembler than you think it is using.

Nado15, 2021-03-11:
Thanks for the advice. I have deleted the old assembler and using the new one. Do I might have to recompile the whole gcc toolchain with my changes or is it really enough when I just replace the assemblers?

closed 2021-03-11 by Nado15

jim-wilson, 2021-03-11:
It is enough to just replace the assembler.