riscv-collab/riscv-gnu-toolchain #865: Is it possible to generate a RISC-V native gcc with --enable-multilib option

billasurendra, 2021-03-18:
I have seen many post regarding gcc cross-compiler/native gcc, here every one is using in their configuration command with --disable-multilib. Except in RISC-V Fedora Image, In this I have checked they configured with --enable-multilib option. 

In RISC-V Fedora image I tried rebuilding gcc without C, but I am unable to do that with --march=rv64imafd   --enable-multilib options.

Even with help of gcc cross compiler, I have tried  to building native gcc with this options --march=rv64imafd --enable-multilib options but i am not succeed.

Can anyone please help how I can generate a native compiler with --enable-multilib.

Please clarify anyone, what is the use of --enable-multilib in gcc configuration ?.

Thanks in advance
Billa

jim-wilson, 2021-03-18:
enable-multilib doesn't work for native builds unless you also use with-multilib-list, which is what Fedora does.

billasurendra, 2021-03-18:
multilib-list is not providing provision for rv64imafd

jim-wilson, 2021-03-18:
> multilib-list is not providing provision for rv64imafd

That doesn't make any sense.  I mentioned "with-multilib-list" not "multilib-list".  --with-multilib-list does not take or use an arch.  It takes an ABI, and with or without C doesn't change the ABI.

If you want help, you must state exactly what you did and exactly what error you got.  If you didn't get an error, but didn't get what you expected, then you must state exactly what you expected and exactly what is different from what you expected.

--enable-multilib does nothing useful for a native compiler build, unless you use it with --with-multilib-list, in which case it changes some of the install paths.  Some linux distros require the install paths that they get with --enable-mutilib for the system compiler, so they build the system compiler with both --enable-multilib and --with-multilib-list.  If you aren't building a system compiler, then don't use --enable-multilib.

billasurendra, 2021-03-19:
Actually my plan is, I wanted to rebuild GCC SRPM (without compressed instruction support) on RISC-V Fedora Linux distro. For that in gcc.spec file I have changed following line (without C option) and I am giving error below of changed one.

%ifarch riscv64
	--with-arch=rv64g  --enable-multilib --with-multilib-list=lp64 \
%endif

Error:

#error "rv32i-based targets are not supported"

TommyMurphyTM1234, 2022-09-29:
No update/activity in over a year and no obvious issue with the RISC-V GCC toolchain or build process itself identified.

closed 2022-09-29 by TommyMurphyTM1234