riscv-collab/riscv-gnu-toolchain #866: can't build in mac

quantrpeter, 2021-03-22:
/bin/sh: flock: command not found

[image]

thanks

jim-wilson, 2021-03-22:
I don't have a mac, so can't directly support builds on mac, other than to suggest that you need homebrew, and you might need to substitute the gnu versions of tools when macos has non-gnu versions that work differently, e.g. use gsed instead of sed, gtar instead of tar, and gawk instead of awk.  homebrew should have a flock program.

sifive/freedom-tools does have support for building on a mac, you might want to look at that.  This builds more than the toolchain, but "make toolchain" on darwin should just build a toolchain.

jim-wilson, 2021-03-22:
You can also try downloading mac binaries if you can't figure out how to build it.  There is a copy of the SiFive toolchain built for macos on the Sifive web site software page, and links to a number of third party toolchains, though I don't know if any of those support macos offhand.  I think the third party ones are mostly MS Windows.

ilg-ul, 2021-03-22:
If you are searching for a newlib toolchain, the **xPack GNU RISC-V Embedded GCC** supports Intel macOS:

- https://github.com/xpack-dev-tools/riscv-none-embed-gcc-xpack/releases

closed 2021-03-23 by quantrpeter