riscv-collab/riscv-gnu-toolchain #867: GDB ELF detection breaks with -Werror=implicit-function-declaration

jrtc27, 2021-03-23:
We fixed this upstream but the riscv-gdb submodule here doesn't contain the fix (b413232211bf7c7754095b017f27774d70646489) yet (and there hasn't been a new upstream release with it yet); could you please cherry-pick it to the riscv fork?

jim-wilson, 2021-03-23:
Unfortunately cherry picking gives me multiple conflicts, so I added the patch by hand, and created riscv-binutils-gdb pr #253.  Verified by building with CFLAGS="-O2 -g -Werror=implicit-function-declaration".  I think this will also help macos builds.

kito-cheng, 2021-03-25:
@jrtc27 https://github.com/riscv/riscv-gnu-toolchain/pull/868 should fix this, feel free to reopen this if it's not work to you.

closed 2021-03-25 by kito-cheng