riscv-collab/riscv-gnu-toolchain #870: RISCV with Vector Extension - Support required for Toolchain and Spike installation

SGR0101, 2021-03-31:
Hi,

**I tried below program with rvv-0.7.x and latest spike as on (30 March 2020)

_SAMPLE CODE_::**

    _0000000080000000 <main>:
    80000000:	300025f3          	csrr	a1,mstatus
    80000004:	2005e593          	ori	a1,a1,512
    80000008:	30059073          	csrw	mstatus,a1
    8000000c:	0010071b          	addiw	a4,zero,1
    80000010:	01f71713          	slli	a4,a4,0x1f
    80000014:	00073783          	ld	a5,0(a4)
    80000018:	008072d7          	vsetvli	t0,zero,e32,m1,d1
    8000001c:	1205e207          	vlw.v	v4,(a1)
    80000020:	0005f557          	vsetvli	a0,a1,e8,m1,d1_

**_In SPIKE::_**

spike_v_ext --isa=RV64GV --varch=vlen:128,elen:16,slen:128 -d mentry
warning: tohost and fromhost symbols not in ELF; can't communicate with target
: 
core   0: 0x0000000000001000 (0x00000297) auipc   t0, 0x0
: 
core   0: 0x0000000000001004 (0x02028593) addi    a1, t0, 32
: 
core   0: 0x0000000000001008 (0xf1402573) csrr    a0, mhartid
: 
core   0: 0x000000000000100c (0x0182b283) ld      t0, 24(t0)
: 
core   0: 0x0000000000001010 (0x00028067) jr      t0
: 
core   0: >>>>  main
core   0: 0x0000000080000000 (0x300025f3) csrr    a1, mstatus
: 
core   0: 0x0000000080000004 (0x2005e593) ori     a1, a1, 512
: 
core   0: 0x0000000080000008 (0x30059073) csrw    mstatus, a1
: 
core   0: 0x000000008000000c (0x0010071b) addiw   a4, zero, 1
: 
core   0: 0x0000000080000010 (0x01f71713) slli    a4, a4, 31
: 
core   0: 0x0000000080000014 (0x00073783) ld      a5, 0(a4)
: 
core   0: 0x0000000080000018 (0x008072d7) vsetvli t0, zero, e16, m1, tu, mu
: 
core   0: 0x000000008000001c (0x1205e207) vle512.v (args unknown)
core   0: **_exception trap_illegal_instruction_**, epc 0x000000008000001c
core   0:           tval 0x000000001205e207

I could successfully compile the code in the link below

https://github.com/sifive/riscv-binutils-gdb/blob/rvv/gas/testsuite/gas/riscv/vector-insns.s

I believe the illegal instruction is due to the different versions of spike and tool chain.

Can any one give the link for _latest stable_ **RISCV tool-chain with vector extension** and matching **Spike** and the **_commands for installation_**. 

Thank You.

jim-wilson, 2021-03-31:
The vector extension has evolved significantly, and had many compatibility breaks.  You need to use a gcc and spike that support exactly the same version of the vector extension.  spike is probably at 1.0 or so, and there is no useful compatbility between 0.7 and 1.0 of the vector spec drafts.

You should try using the rvv-intrinsic branch instead of rvv-0.7.x.  The rvv-0.7.x branch is obsolete and no longer supported.  The rvv-intrinsic branch should support rvv 1.0.  Also note that the GCC V port is incomplete and not actively maintained, so there is no guarantee that it will work correctly in all cases.  Current V development work is focused on LLVM.

SGR0101, 2021-04-01:
Sir, I tried the way as you suggested...Now compilation and Spike both worked... Thank you !!!

Leaving here the steps to install riscv vector ext tool-chain on UBUNTU as on 01/April/2021.

1. **git clone https://github.com/riscv/riscv-gnu-toolchain.git --branch rvv-intrinsic**

2. **sudo apt-get install autoconf automake autotools-dev curl python3 libmpc-dev libmpfr-dev libgmp-dev gawk build-essential bison flex texinfo gperf libtool patchutils bc zlib1g-dev libexpat-dev**

3. **git submodule update --init --recursive --depth 1 riscv-binutils riscv-gcc riscv-glibc riscv-dejagnu riscv-newlib riscv-gdb**

   (NOT SURE IF ABOVE STEP IS NECESSARY)

4. cd riscv-gnu-toolchain

5. **./configure --prefix=/home/your/install_path_for_toolchain/your_dir_name --with-arch=rv64gcv --with-abi=lp64d**

6. **make** 

This will take up your time....

8. Finally add the PATH to .bashrc file and save it, as shown below.

export PATH=/home/your/install_path_for_toolchain/your_dir_name/bin:$PATH
export LD_LIBRARY_PATH=/home/your/install_path_for_toolchain/your_dir_name/riscv64-unknown-elf/lib:$LD_LIBRARY_PATH

9. Take a new terminal and type command below.
      **which riscv64-unknown-elf-gcc**

If the above command returns your installed path to terminal as below, your tool chain is ready for compilation
**/home/your/install_path_for_toolchain/your_dir_name/bin/riscv64-unknown-elf-gcc**

Similarly you can follow steps in Github to install the latest Spike....

ThankYou.

closed 2021-04-01 by SGR0101