riscv-collab/riscv-gnu-toolchain #871: Status of the rvv-intrinsic branch

mp-17, 2021-03-31:
Hello everyone,

First of all, thanks for your great work on the toolchain.
I successfully compiled a program with RVV intrinsics using the `rvv-intrinsic` toolchain, and then I run the binary on SPIKE (RVV 0.10). Thus, I guess `rvv-intrinsic` targets the current RVV 0.10. 
Do you approximately know which is the current status of the branch `rvv-intrinsic` with respect to the specs and for what concerns its stability?

Thanks a lot,
Matteo

jim-wilson, 2021-03-31:
The rvv-intrinsic branch is not actively maintained.  Current focus for V compiler work is on LLVM.

mp-17, 2021-03-31:
Thanks a lot, Jim! 
> Current focus for V compiler work is on LLVM.

Is this only related to the RVV intrinsics or it is true for RVV in general?

jim-wilson, 2021-03-31:
RVV work in general is focused on LLVM at the moment.

kito-cheng, 2021-04-01:
RVV intrinsic for GCC will become community work and sort of my side project, but I believe the implementation is stable enough to use, and it's the relative stable open source implementation you could get, RVV intrinsic for upstream clang/LLVM still incomplete, so you might still need wait another couple of month for clang/LLVM.

TommyMurphyTM1234, 2022-09-29:
The `rvv-next` branch (https://github.com/riscv-collab/riscv-gnu-toolchain/tree/rvv-next) is where work to implement/enhance support for the V extension v1.0 is being carried out these days.

closed 2022-09-29 by TommyMurphyTM1234