riscv-collab/riscv-gnu-toolchain #874: Separate build and install steps

dhouck, 2021-04-05:
The build user should not need access to the prefix directory just to build the project; this should only be necessary when installing to that directory.  It's a bad idea to run the entire build as root just because the final install needs to be done to a root-owned directory.

That is, the error message `Sorry, you don't have permission to write to '/opt/riscv', use --prefix to specify another path, or use 'sudo make' if you *REALLY* want to install into '/opt/riscv'` should not be given; there's a reason most programs separate out the `make` and `make install` step.

jim-wilson, 2021-04-05:
riscv-gnu-toolchain is not a single program.  It is a collection of multiple programs from multiple sources, which limits what we can do with it.  The GCC build needs binutils, so binutils must be installed before the gcc build starts.  The glibc build needs gcc, so gcc must be installed before the glibc build starts.  Etc.  Separating the build and install steps is simply not practical.  

I do agree that you should not do the build as root.  You can install someplace else and copy into /opt/riscv, or  make /opt/riscv temporiarly writable to the user doing the build, or install someplace that the user already has write access to, etc.

riscv-gnu-toolchain is maintained by volunteers, and there is far more work to do than volunteers to do it, so some problems like this one may never be fixed.

dhouck, 2021-04-06:
Thanks for the quick response!

I was pretty sure that `gcc` at least stored its configured path and the solution of just copying it in wouldn't work.  Temporarily making it writeable could, as could various other solutions of increasing complexity.

I get that this is difficult (I'm pretty sure gcc at least offers a way around that issue but I'm not sure since I've only built it as a non-cross compiler before and not recently).  Would it be possible to at least change the error message to suggest the temporarily-writeable solution instead of `sudo make`?

TommyMurphyTM1234, 2022-09-29:
My understanding is that the separation of build and install for the riscv-gnu-toolchain makefile is not likely to happen for the reasons explained by @jim-wilson above and, as such, I'm inclined to close this rather than leave it open even as an enhancement issue. @cmuellner - what do you think?

cmuellner, 2022-09-29:
I fully agree, that we should not encourage people to invoke `sudo make`.
How do people like PR #1134?

TommyMurphyTM1234, 2022-09-29:
> I fully agree, that we should not encourage people to invoke `sudo make`. How do people like PR #1134?

Thanks @cmuellner. LGTM.

closed 2022-09-29 by TommyMurphyTM1234