riscv-collab/riscv-gnu-toolchain #876: gfortran compilation error

amithmath, 2021-04-07:
I am  trying to generate fortan compiler for embedded elf. I modified Makefile: https://github.com/riscv/riscv-gnu-toolchain/blob/master/Makefile.in at line 623 as:

--enable-languages=c,c++,fortran \

but I am getting compilation error: 

checking whether the c compiler works... no

am I missing something?

Thanks.

amithmath, 2021-04-07:
more detailed error:

configure: error: C compiler cannot create executables
See `config.log' for more details

jim-wilson, 2021-04-07:
Where exactly did you get the error?  riscv-gnu-toolchain builds many different things, and any one of them could have failed.  It sounds like an error from a configure script, which means you should check the config.log file in the directory where the configure failed to see what the configure error is.

amithmath, 2021-04-07:
Following are config log and configure scripts

config log
-------
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by riscv-toolchain configure 1.0, which was
generated by GNU Autoconf 2.69.  Invocation command line was

  $ ./configure --prefix=/home/centos/M2/M2_CORE/M2_multi_C/software/riscv-tools/riscv-install --disable-linux --with-arch=rv32imaf --with-abi=ilp32f

## --------- ##
## Platform. ##
## --------- ##

hostname = ip-172-31-56-86.ec2.internal
uname -m = x86_64
uname -r = 3.10.0-957.27.2.el7.x86_64
uname -s = Linux
uname -v = #1 SMP Mon Jul 29 17:46:05 UTC 2019

/usr/bin/uname -p = x86_64
/bin/uname -X     = unknown

/bin/arch              = x86_64
/usr/bin/arch -k       = unknown
/usr/convex/getsysinfo = unknown
/usr/bin/hostinfo      = unknown
/bin/machine           = unknown
/usr/bin/oslevel       = unknown
/bin/universe          = unknown

PATH: /home/centos/M2/M2_CORE/M2_multi_C/software/riscv-tools/depends/bin
PATH: /home/centos/M2/M2_CORE/M2_multi_C/software/riscv-tools/depends/bin
PATH: /opt/Xilinx/SDx/2019.1.op2552052/bin
PATH: /opt/Xilinx/Vivado/2019.1.op2552052/bin
PATH: /usr/local/bin
PATH: /usr/bin
PATH: /usr/local/sbin
PATH: /usr/sbin
PATH: /srv/git/centos-git-common
PATH: /home/centos/.local/bin
PATH: /home/centos/bin

## ----------- ##
## Core tests. ##
## ----------- ##

configure:1895: checking for gcc
configure:1922: result: gcc
configure:2151: checking for C compiler version
configure:2160: gcc --version >&5
gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-44)
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

configure:2171: $? = 0
configure:2160: gcc -v >&5
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/4.8.5/lto-wrapper
Target: x86_64-redhat-linux
Configured with: ../configure --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-bootstrap --enable-shared --enable-threads=posix --enable-checking=release --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-linker-hash-style=gnu --enable-languages=c,c++,objc,obj-c++,java,fortran,ada,go,lto --enable-plugin --enable-initfini-array --disable-libgcj --with-isl=/builddir/build/BUILD/gcc-4.8.5-20150702/obj-x86_64-redhat-linux/isl-install --with-cloog=/builddir/build/BUILD/gcc-4.8.5-20150702/obj-x86_64-redhat-linux/cloog-install --enable-gnu-indirect-function --with-tune=generic --with-arch_32=x86-64 --build=x86_64-redhat-linux
Thread model: posix
gcc version 4.8.5 20150623 (Red Hat 4.8.5-44) (GCC) 
configure:2171: $? = 0
configure:2160: gcc -V >&5
gcc: error: unrecognized command line option '-V'
[... 4799 lines cut ...]
# config.status does its own redirection, appending to config.log.
# Unfortunately, on DOS this fails, as config.log is still kept open
# by configure, so config.status won't be able to write to it; its
# output is simply discarded.  So we exec the FD to /dev/null,
# effectively closing config.log, so it can be properly (re)opened and
# appended to by config.status.  When coming back to configure, we
# need to make the FD available again.
if test "$no_create" != yes; then
  ac_cs_success=:
  ac_config_status_args=
  test "$silent" = yes &&
    ac_config_status_args="$ac_config_status_args --quiet"
  exec 5>/dev/null
  $SHELL $CONFIG_STATUS $ac_config_status_args || ac_cs_success=false
  exec 5>>config.log
  # Use ||, not &&, to avoid exiting from the if with $? = 1, which
  # would make configure fail if this is the last instruction.
  $ac_cs_success || as_fn_exit 1
fi
if test -n "$ac_unrecognized_opts" && test "$enable_option_checking" != no; then
  { $as_echo "$as_me:${as_lineno-$LINENO}: WARNING: unrecognized options: $ac_unrecognized_opts" >&5
$as_echo "$as_me: WARNING: unrecognized options: $ac_unrecognized_opts" >&2;}
fi

---------

amithmath, 2021-04-07:
[config.log](https://github.com/riscv/riscv-gnu-toolchain/files/6268956/config.log)

[configure.txt](https://github.com/riscv/riscv-gnu-toolchain/files/6268957/configure.txt)

amithmath, 2021-04-07:
above are the config log and configure scripts

amithmath, 2021-04-07:
As per the config.log error is at line#116-118

amithmath, 2021-04-07:
checking whether the C compiler works... no
configure: error: in `/home/centos/M2/M2_CORE/M2_multi_C/software/riscv-tools/riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv32-unknown-elf-dramfs/libbacktrace':
configure: error: C compiler cannot create executables
See `config.log' for more details
make[5]: *** [configure-target-libbacktrace] Error 1
make[5]: Leaving directory `home/centos/M2/M2_CORE/M2_multi_C/software/riscv-tools/riscv-gnu-toolchain/build-gcc-newlib-stage2'
make[4]: *** [all] Error 2
make[4]: Leaving directory `home/centos/M2/M2_CORE/M2_multi_C/software/riscv-tools/riscv-gnu-toolchain/build-gcc-newlib-stage2'
make[3]: *** [stamps/build-gcc-newlib-stage2] Error 2
make[3]: Leaving directory `home/centos/M2/M2_CORE/M2_multi_C/software/riscv-tools/riscv-gnu-toolchain'
make[2]: *** [build-riscv-gnu-tools] Error 2
make[2]: Leaving directory `home/centos/M2/M2_CORE/M2_multi_C/software/riscv-tools'
make[1]: *** [build-all] Error 2
make[1]: Leaving directory `/home/centos/M2/M2_CORE/M2_multi_C/software/riscv-tools'
make: *** [tools] Error 2

jim-wilson, 2021-04-07:
The config.log you uploaded is from the top level directory.  Every package in riscv-gnu-toolochain has its own configure script, and will create its own config.log file, so there are several dozen of them in the build.  You need the one in the dir where the configure failed.  See the path with libbacktrace in your build log.

But looking at the config.log that you did upload, I see that you are missing gmp, mpfr, and mpc.  You will need these for fortran support.  The linux build rule has a step that runs download_prerequisites to add the packages if you are missing them.  You will also need this for newlib to build fortran support.  I think this is unrelated to your libbacktrace failure though.

amithmath, 2021-04-07:
I have attached config.log file in libbacktrace directory
[config.log](https://github.com/riscv/riscv-gnu-toolchain/files/6273276/config.log)

jim-wilson, 2021-04-07:
I see
/home/centos/M2/M2_CORE/M2_multi_C/software/riscv-tools/riscv-install/riscv32-unknown-elf-dramfs/bin/ld: cannot find crt0.o: No such file or directory
Maybe the newlib build failed?  crt0.o comes from libgloss which is part of the newlib build.  There should be a copy in $prefix/$target/lib.  The failing gcc command does have
-B/home/centos/M2/M2_CORE/M2_multi_C/software/riscv-tools/riscv-install/riscv32-unknown-elf-dramfs/lib/
so if crt0.o was there gcc should have been able to find it.  That suggests that the newlib build failed.

Or maybe there is some other more serious problem.  There is probably a reason why we don't build fortran for the embedded elf newlib target, but I don't what it is offhand.  There might be other fixes required to make this work.

amithmath, 2021-04-08:
there is crt0.o in /home/centos/M2/M2_CORE/M2_multi_C/software/riscv-tools/riscv-install/riscv32-unknown-elf-dramfs/lib/
please tell me how to get it?

amithmath, 2021-04-08:
there is no crt0.o in /home/centos/M2/M2_CORE/M2_multi_C/software/riscv-tools/riscv-install/riscv32-unknown-elf-dramfs/lib/
please tell me how to get it?

amithmath, 2021-04-08:
It seems newlib build failed.

amithmath, 2021-04-08:
I did a small trick, ./configure command line options I gave --enable-languages=c,c++, fortran along with other options. It successfully compiled and generated crt0.o as well but gfortran binary not generated in bin directory as shown in the picture?
[image]

amithmath, 2021-04-08:
I did a small trick, ./configure command line options I gave --enable-languages=c,c++, fortran along with other options. It successfully compiled and generated crt0.o as well but gfortran binary not generated in bin directory as shown below
[image]
 in the picture?

jim-wilson, 2021-04-08:
Which configure did you add --enable-languages?  We don't support that at the top level configure.  That will only work with the riscv-gcc configure.  So you still need to modify the build-gcc-newlib-stage2 rule like you did before, and you probably still need the download_prerequisites change I mentioned before.  The newlib build failure may be unrelated to the fortran build problems.

amithmath, 2021-04-09:
Yes  --enable-languages. Please let me know to the command to download 'download_prerequisites'? I have to download in risc-v-gnu-toolchain?

jim-wilson, 2021-04-09:
It should be run automatically from the Makefile if you need it.  See the build-gcc-newlib-stage1 rule for instance.  Are you still gettng an error with the fortran build?  The problem I looked at earlier might have been a cascading error, or I might have gotten confused by looking at the wrong config.log file.  The real problem was the broken newlib build.  Now that is fixed, you might be running into a different problem that you haven't reported yet.

amithmath, 2021-04-09:
Yes I am getting fortran build error, if I  remove --enable-languages=c,c++,fortran at ./configure  and add --enable-languages=c,c++,fortran  at line 623: https://github.com/riscv/riscv-gnu-toolchain/blob/master/Makefile.in

I just did the following:

cd software/riscv-tools/riscv-gnu-toolchain/riscv-gcc
./contrib/download_prerequisites 

I started compiling....

amithmath, 2021-04-09:
I did modify Makefile as per your suggestions for build-gcc-newlib-stage2 to add 'download_prerequisites'.  I am still getting same error. Following is the confg.log
[config.log](https://github.com/riscv/riscv-gnu-toolchain/files/6282836/config.log)

amithmath, 2021-04-09:
In order to get rid off 'newlib build failure', I enabled fortran (--enable-languages=c,c++, fortran)  both build-gcc-newlib-stage1 and build-gcc-newlib-stage2, it successfully compiled but not generated gfortran binary.  I then enabled '--enable-languages=c,c++, fortran' only at build-gcc-newlib-stage2, it is successfully compiled but not generated gfortran binary. Just wondering what's happening?

amithmath, 2021-04-09:
of course I copied the line#517 at: https://github.com/riscv/riscv-gnu-toolchain/blob/master/Makefile.in to after line#614 at https://github.com/riscv/riscv-gnu-toolchain/blob/master/Makefile.in

amithmath, 2021-04-10:
Please help me this issue, I am badly stuck....Many thanks.

amithmath, 2021-04-11:
Thanks. After so many trials, I successfully built gfortran

amithmath, 2021-04-30:
I am trying to compile simple fortran hello world program I am getting following error:  
fatal error: cannot read spec file 'libgfortran.spec': No such file or directory

I used this command to compile:  path/where riscv32-elf bin installed/gfortran hello_word.f90

I am wondering, what could be the problem?

amithmath, 2021-04-30:
I did: 'find -iname libgfortran.spec' but I could not get any files....

amithmath, 2021-04-30:
I am using gcc version 9.0.2

amithmath, 2021-04-30:
I am using gcc version 9.2.0

closed 2021-04-30 by amithmath