riscv-collab/riscv-gnu-toolchain #877: stdint.h: No such file or directory

QiaowenYoung, 2021-04-07:
I am using riscv64-unknown-linux-gnu, and got the following issue:

`riscv64-unknown-linux-gnu/10.2.0/include/stdint.h:9:16: fatal error: stdint.h: No such file or directory
    9 | # include_next <stdint.h>
      |                ^~~~~~~~~~
`
Anyone knows how to fix this?

jim-wilson, 2021-04-07:
There should be a stdint.h file in $prefix/sysroot/usr/include/stdint.h.  If it isn't there, then your build failed before it finished and you didn't notice the error.  If you have the build log, go back and look for an error.  If you have the build tree, try running make again.

If you are on a modern linux distro, you can just apt/dnf/yum install a riscv64-linux toolchain.

QiaowenYoung, 2021-04-08:
Tried and succeeded! Thank you very much.

closed 2021-04-08 by QiaowenYoung