riscv-collab/riscv-gnu-toolchain #878: riscv64-unknown-elf-ld: -march=: ISA string must begin with rv32 or rv64 when vector extension 0.10 is enabled

alextsao1999, 2021-04-08:
Hi there,
I was trying to compile a little case with clang and then use riscv-gnu-toolchain to link them, but the linker prompts the following error.I don’t know if my version or compilation options are wrong.

Here is the commands

compile riscv-gnu-toolchain
```
../configure --prefix=$RISCV --with-arch=rv64gcv0p10
make
```

compile my case
```
clang --target=riscv64-unknown-elf --sysroot=$RISCV/riscv64-unknown-elf --gcc-toolchain=$RISCV -march=rv64gcv0p10 -menable-experimental-extensions -O3 hello.c -o hello -v
```

```
clang version 13.0.0 (https://gitee.com/mirrors/llvm-project.git c8e56f394af0b9e32c413d62a0e7aebbba3e6b70)
Target: riscv64-unknown-unknown-elf
Thread model: posix
InstalledDir: /home/root/llvm/build/bin
Found candidate GCC installation: /home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0
Selected GCC installation: /home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0
 "/home/root/llvm/build/bin/clang-13" -cc1 -triple riscv64-unknown-unknown-elf -emit-obj --mrelax-relocations -disable-free -main-file-name hello.c -mrelocation-model static -mframe-pointer=none -fmath-errno -fno-rounding-math -mconstructor-aliases -nostdsysteminc -target-feature +m -target-feature +a -target-feature +f -target-feature +d -target-feature +c -target-feature +experimental-v -target-feature +experimental-zvamo -target-feature +experimental-zvlsseg -target-feature +relax -target-feature -save-restore -target-abi lp64d -msmall-data-limit 8 -debugger-tuning=gdb -v -fcoverage-compilation-dir=/home/root/build -resource-dir /home/root/llvm/build/lib/clang/13.0.0 -isysroot /home/root/install/rvv/riscv64-unknown-elf -internal-isystem /home/root/install/rvv/riscv64-unknown-elf/include -O3 -fdebug-compilation-dir=/home/root/build -ferror-limit 19 -fno-signed-char -fgnuc-version=4.2.1 -fcolor-diagnostics -vectorize-loops -vectorize-slp -faddrsig -o /tmp/hello-29ad11.o -x c hello.c
clang -cc1 version 13.0.0 based upon LLVM 13.0.0git default target x86_64-unknown-linux-gnu
#include "..." search starts here:
#include <...> search starts here:
 /home/root/install/rvv/riscv64-unknown-elf/include
 /home/root/llvm/build/lib/clang/13.0.0/include
End of search list.
 "/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld" --sysroot=/home/root/install/rvv/riscv64-unknown-elf -m elf64lriscv /home/root/install/rvv/riscv64-unknown-elf/lib/crt0.o /home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/crtbegin.o -L/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0 -L/home/root/install/rvv/riscv64-unknown-elf/lib /tmp/hello-29ad11.o --start-group -lc -lgloss --end-group -lgcc /home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/crtend.o -o hello
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: -march=rv64i2p0_m2p0_a2p0_f2p0_d2p0_c2p0_v0p10_zvamo0p10_zvlsseg0p10: Invalid or unknown z ISA extension: 'zvamo'
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /tmp/hello-29ad11.o
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: -march=: ISA string must begin with rv32 or rv64
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /home/root/install/rvv/riscv64-unknown-elf/lib/libc.a(lib_a-exit.o)
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: -march=: ISA string must begin with rv32 or rv64
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /home/root/install/rvv/riscv64-unknown-elf/lib/libc.a(lib_a-impure.o)
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: -march=: ISA string must begin with rv32 or rv64
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /home/root/install/rvv/riscv64-unknown-elf/lib/libc.a(lib_a-init.o)
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: -march=: ISA string must begin with rv32 or rv64
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /home/root/install/rvv/riscv64-unknown-elf/lib/libc.a(lib_a-memset.o)
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: -march=: ISA string must begin with rv32 or rv64
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /home/root/install/rvv/riscv64-unknown-elf/lib/libc.a(lib_a-puts.o)
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: -march=: ISA string must begin with rv32 or rv64
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /home/root/install/rvv/riscv64-unknown-elf/lib/libc.a(lib_a-strlen.o)
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: -march=: ISA string must begin with rv32 or rv64
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /home/root/install/rvv/riscv64-unknown-elf/lib/libc.a(lib_a-__call_atexit.o)
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: -march=: ISA string must begin with rv32 or rv64
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /home/root/install/rvv/riscv64-unknown-elf/lib/libc.a(lib_a-findfp.o)
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: -march=: ISA string must begin with rv32 or rv64
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /home/root/install/rvv/riscv64-unknown-elf/lib/libc.a(lib_a-fini.o)
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: -march=: ISA string must begin with rv32 or rv64
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /home/root/install/rvv/riscv64-unknown-elf/lib/libc.a(lib_a-fvwrite.o)
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: -march=: ISA string must begin with rv32 or rv64
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /home/root/install/rvv/riscv64-unknown-elf/lib/libc.a(lib_a-fwalk.o)
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: -march=: ISA string must begin with rv32 or rv64
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /home/root/install/rvv/riscv64-unknown-elf/lib/libc.a(lib_a-mallocr.o)
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: -march=: ISA string must begin with rv32 or rv64
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /home/root/install/rvv/riscv64-unknown-elf/lib/libc.a(lib_a-memchr.o)
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: -march=: ISA string must begin with rv32 or rv64
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /home/root/install/rvv/riscv64-unknown-elf/lib/libc.a(lib_a-memcpy.o)
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: -march=: ISA string must begin with rv32 or rv64
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /home/root/install/rvv/riscv64-unknown-elf/lib/libc.a(lib_a-memmov [... cut ...]
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /home/root/install/rvv/riscv64-unknown-elf/lib/libc.a(lib_a-errno.o)
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: -march=: ISA string must begin with rv32 or rv64
/home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../bin/riscv64-unknown-elf-ld: failed to merge target specific data of file /home/root/install/rvv/lib/gcc/riscv64-unknown-elf/10.2.0/crtend.o
clang-13: error: ld command failed with exit code 1 (use -v to see invocation)
```

kito-cheng, 2021-04-09:
> -march=rv64i2p0_m2p0_a2p0_f2p0_d2p0_c2p0_v0p10_zvamo0p10_zvlsseg0p10: Invalid or unknown z ISA extension: 'zvamo'

Your linker didn't recognize zvamo extension, I would suggest you using `rvv-intrinsic` branch[1] to build riscv gnu toolchain and then combine with clang to use.

[1] https://github.com/riscv/riscv-gnu-toolchain/tree/rvv-intrinsic 

@Nelson1225 maybe we could relax the extension name checking during link time?

alextsao1999, 2021-04-09:
Thanks, kito .The problem has been resolved.

closed 2021-04-09 by alextsao1999

Nelson1225, 2021-04-12:
> Your linker didn't recognize zvamo extension, I would suggest you using `rvv-intrinsic` branch[1] to build riscv gnu toolchain and then combine with clang to use.
>
> [1] [...]

Noted, this sounds reasonable.  I will take care of this in the future patches.