riscv-collab/riscv-gnu-toolchain #879: Toolchain cannot call library functions

hqw-lantian, 2021-04-11:
Hello, regarding the use of the RISC-V GCC tool chain, we have encountered the following problems:
When toolchains calls library functions, although the compilation is successful, the generated binary instructions will generate a lot of "space". After adding "space", the instructions are aligned, but after removing the "space", the instructions cannot be aligned, which makes SoC unable to parse the instructions. How to solve the problem. 
THANK YOU!

hqw-lantian, 2021-04-11:
Correct waveform and binary file
[image]
[image]

Wrong waveform and binary file
[image]
[image]

jim-wilson, 2021-04-11:
"space" isn't a well defined technical term.  Are you talking about null bytes?  Nops?  ASCII space characters?  I don't know what your waveform and binary files are trying to show.  I suggest you try using objdump to disassemble.

hqw-lantian, 2021-04-12:
Hello, after calling the library function random function, through disassembly, you can see that the srand function has been successfully called, but the original normal binary instruction will insert a blank in the middle, as shown in my screenshot, lines 11624 and 11641 Two binary instructions 0x00, 0x2D, nothing in between. But when my SoC reads instructions, it ignores these blanks. This also causes that if blanks are counted as specific lines, our binary instruction lines are multiples of 8. If we skip them, our binary instructions will Not a multiple of 8. Therefore, I read the instruction using 64bit as the basic unit, and read the last set of data will appear in the x state, causing the CPU to not run. Hope to get your help

[image]
[image]

hqw-lantian, 2021-04-12:
My risc-v core is BOOMv2, and the tool chain used is --enable-multilib --with-abi=lp64 --with-arch=rv64g.thank you for your reply!

kito-cheng, 2021-04-12:
It's hard to observe the relation between first and second screenshot, but I guess you mean the blank lines in the second screenshot, it seems like extracted from the executable binary by some script, I would suggest you debug that script since the output of toolchain seems correct.

jim-wilson, 2021-04-12:
I don't know what the waveform file is.  We aren't hardware people.

Maybe they are having trouble with an ELF to bin conversion?  If that is the case, then I would suggest trying 
[https://github.com/sifive/freedom-elf2hex](url)

hqw-lantian, 2021-05-07:
The problem is solved, thank you for your help

TommyMurphyTM1234, 2022-09-27:
Not sure what the issue was but @hqw-lantian says that it was resolved.