riscv-collab/riscv-gnu-toolchain #880: Problem in machine code in Hex file for "flw f14, 8(x2)"

abdulhameed-lm, 2021-04-12:
My core's Base ISA is RV32IMFC. Whenever I try Floating Load or Store instruction with Base Address stored in X2 register, my program.hex is generated with wrong machine code. For "flw f14, 8(x2)" my machine code in Program.hex is "6722", while it should have been "00812707". Same goes for all floating load and stores with base integer register X2. These instruction work fine with other 31 registers except x2. My march and mabi is set to following: -mabi=ilp32f -march=rv32imafc. Can you guide me, what am i doing wrong?

aswaterman, 2021-04-12:
Note the "c" in "rv32imafc": you are enabling the compressed ISA, and your
flw is being compressed into c.flwsp.  Things are working as expected.

abdulhameed-lm, 2021-04-12:
oh now i understand, okay thanks

closed 2021-04-12 by abdulhameed-lm