riscv-collab/riscv-gnu-toolchain #881: PyTorch

amithmath, 2021-04-15:
I am wondering, does RISC-V gcc compiler support calling PyTorch scripts in C codes? I can call Python scripts in C by just including Python.h header file and using Python C APIs. If I do the same for same for calling PyTorch script, I am getting following error:

import torch
ModuleNotFoundError: No module named 'torch'

dlan17, 2021-04-15:
I bet you should install pyTorch[1] package first? not sure if it need any additional work to port to RISC-V ..

[1] https://pytorch.org/

dlan17, 2021-04-15:
I'm not familiar with pyTorch..

As my understanding, you need to install pyTorch and the dependencies it requires, you probably could take a look at Gentoo's ebuild for example[1], may get some ideas

[1] https://gitweb.gentoo.org/proj/sci.git/tree/sci-libs/pytorch/pytorch-1.7.1-r2.ebuild

amithmath, 2021-04-16:
OK thanks.....

closed 2021-04-16 by amithmath