riscv-collab/riscv-gnu-toolchain #882: If you have a target linux system that already has a glibc build, then it is fairly easy to do a toolchain build using a sysroot.  Just create a sysroot that contains /lib*, /usr/lib*, and /usr/include from the target root filesystem.  Or alternatively, if your target root file system is mounted on your build system, you can use that directly.

billasurendra, 2021-04-16:
If you have a target linux system that already has a glibc build, then it is fairly easy to do a toolchain build using a sysroot.  Just create a sysroot that contains /lib*, /usr/lib*, and /usr/include from the target root filesystem.  Or alternatively, if your target root file system is mounted on your build system, you can use that directly.

If you don't have a target linux system with a glibc build, then a toolchain build gets very complicated because of interdependencies between kernel/glibc/binutils/gcc which requires multiple partial builds before you can do final full builds.  It is best to use a script that someone has written.  Every linux distro has their own scripts for this.  OpenEmbedded/Yocto are popular choices for embedded linux.  There is also crosstools-ng that I have used but I'm not sure if it supports a canadian cross build.  We have the linux-native rule in our makefiles, but we don't use it, and offer no guarantees that it will work.

Here are the scripts I used last time I canadian cross built a native compiler using fedora header files.  You need to build the cross (x) binutils and gcc first, then the host (h) binutils and gcc.

```
gamma02:2007$ more x-binutils.script 
../binutils-2.30/configure --target=riscv64-unknown-linux-gnu --prefix=/scratch/
jimw/fedora-cross/install --with-sysroot=/scratch/jimw/fedora-cross/sysroot
gamma02:2008$ more x-gcc.script 
../gcc-7.3.0/configure --target=riscv64-unknown-linux-gnu --prefix=/scratch/jimw
/fedora-cross/install --with-sysroot=/scratch/jimw/fedora-cross/sysroot --disabl
e-multilib
gamma02:2009$ more h-binutils.script 
../binutils-2.30/configure --host=riscv64-unknown-linux-gnu --target=riscv64-unk
nown-linux-gnu --prefix=/scratch/jimw/fedora-cross/h-install --with-sysroot=/scr
atch/jimw/fedora-cross/sysroot
gamma02:2010$ more h-gcc.script 
../gcc-7.3.0/configure --host=riscv64-unknown-linux-gnu --target=riscv64-unknown
-linux-gnu --prefix=/scratch/jimw/fedora-cross/h-install --with-sysroot=/scratch
/jimw/fedora-cross/sysroot --disable-multilib
gamma02:2011$ 
```

_Originally posted by @jim-wilson in https://github.com/riscv/riscv-gnu-toolchain/issues/333#issuecomment-382492988_

billasurendra, 2021-04-16:
Hi Jim,

I have a doubt in RISC-V native compiler building Process. Here my target image having folders like:

**bin   boot   bootgrub   dev   etc   home   lib   lib64   media   mnt   opt   proc   root   sbin   srv   sys   temp   usr   var** 

As you suggested in above issue. How to create a sysroot ?.  Here really I am not understanding where I have to create sysroot  that contains /lib*, /usr/lib*, and /usr/include from the target root filesystem. When I am running  x-binutils.script, what path I have to give in this place **--with-sysroot=/scratch/jimw/fedora-cross/sysroot**.

FIY, I have target root file system on my host pc itself.

Can you please clarify my doubt and give me a path to building a native compiler.

Thanks in advance
Billa

jim-wilson, 2021-04-16:
The target has a file system.  The build machine has a file system.  You need to copy files from the target to the build machine.  This is the sysroot.  It can be placed anywhere on the build machine.  This doesn't matter.  The sysroot needs to contain all of the files that the compiler needs, so it needs to contain /lib*, /usr/lib*, and /usr/include.  It can optionally include the entire target filesystem, but if you have a lot of stuff on the target that is a waste of disk space on the build machine.  When you build the cross compiler you point the --with-sysroot= option at the path to the sysroot on the build machine.  So for instance $sysroot/usr/include/stdio.h should be the target stdio.h file.

billasurendra, 2021-04-21:
Hi Jim,

I have followed your suggestions and executed these four scrips in this order  1. x-binutils   -->    2. x-gcc   -->   3. h-binutils    -->   4. h-gcc. For every individual script I have maintained separate folder for result (means  make install). i.e 

1. x_binutils_install  (it is having x86 binaries and libraries)
2. x_gcc_install  (it is having x86 binaries and libraries)
3. h_binutils_install (it is having riscv binaries and libraries)
4. h_gcc_install  (it is having riscv binaries and libraries)

Can you please clarify some of my doubts.

1. I need to copy only h_binutils_install  and h_gcc_install folders only right ?
2.  where I have to copy these two folders in target file system ?
3.  or anything I have missed ?

Thanks in-advance
Billa

jim-wilson, 2021-04-21:
You only copy the h_binutils_install ande h_gcc_install.  You can copy them anywhere.  GCC will probably work better if installed to the same dir on target that it was installed to on the host.  Otherwise, you may have search path issues.  But the search path issues can be fixed by creating a specs file in your gcc install dir and editing it to adding -B or -I or -L options to point to the right dirs.  Once you have the h_gcc_install on the target and working for a hello world program, you should use it to build binutils and gcc natively.  This will get a toolchain with the right paths.

billasurendra, 2021-04-22:
I am getting error while running sample.c
#gcc -o out sample.c
gcc:error trying to exec 'cc1':execvp:No such file or directory

copied cc1 file to same directory where I am running sample.c program.
Can you please help me to solve this error.

jim-wilson, 2021-04-22:
Is this your canadian cross built compiler now running natively?  I mentioned that the paths may not be right.  You can use "gcc --print-search-dirs" to see the paths that it using to find libraries, header files, and subprograms like cc1.

The easiest way to fix this is probably to just create dirs and symlinks on the RISC-V machine so that the paths originally used on the build machine work.

There is a way to fix this by adding special configure options during the canadian cross build process, but I never bothered to learn how to do that.  You would have to read gcc install docs and figure it out yourself.

Another way to fix this is to add options to every gcc run.  If you add a -B<dir>/ option pointing to the dir where the compiler is installed, the compiler should be able to find everything.  This needs to be pointing to the dir where cc1 is installed.  If this doesn't work by itself, then you might also need to add -I and/or -L options.  One way to do this is to create a specs file with -dumpspecs, put it someplace gcc will find it, and then edit the specs file to add the -B option.  Or you can try creating a shell script wrapper and put it on your path.  Etc.

Once you get the canadian cross built gcc working, you should then use it to build another native compiler.  This one will then have all of the paths correct and won't need any workarounds.

billasurendra, 2021-04-27:
Sorry I don't know what is mean by canadian cross built compiler, In my host pc I have riscv cross compilers with that cross compiled binutils and gcc source code. as you suggested I have copied target image content to host and created sysroot and executed x-binutils, x-gcc, h-binutils and h-gcc sripts. finally I have copied h_binutils_install and h_gcc_install folders to target image. 

I have checked **$ gcc --print-search-dirs** in target riscv image, In this:

**install:** /usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.1/ **(This path exist in target image)**

**programs:** =/usr/bin/../libexec/gcc/riscv64-unknown-linux-gnu/7.3.1/: **(This path does not exist in target image)**
           /usr/bin/../libexec/gcc/:/usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64- unknown-linux-gnu/bin/riscv64-unknown-linux-gnu/7.3.1/:  **(This path does not exist in target image)**
           /usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/bin/  **(This path does not exist in target image)**

**libraries:** =/usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.1/:   **(This path exist in target image)**
            /usr/bin/../lib/gcc/:     **(This path exist in target image)**
            /usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/lib/riscv64-unknown-linux-gnu/7.3.1/:    **(This path does not exist in target image)**
            /usr/bin/../lib/gcc/riscv64-unknown-linux-gnu/7.3.1/../../../../riscv64-unknown-linux-gnu/lib/:  **(This path does not exist in target image)** 
            /home/surendrab/MDP/old_v_distro/lj-os/lib64/lp64d/riscv64-unknown-linux-gnu/7.3.1/:  **(not exist)**
            /home/surendrab/MDP/old_v_distro/lj-os/lib64/lp64d/:                                  **(not exist)**
            /home/surendrab/MDP/old_v_distro/lj-os/usr/lib64/lp64d/riscv64-unknown-linux-gnu/7.3.1/: **(not exist)**
            /home/surendrab/MDP/old_v_distro/lj-os/usr/lib64/lp64d/:      **(not exist)**
            /home/surendrab/MDP/old_v_distro/lj-os/lib/riscv64-unknown-linux-gnu/7.3.1/:   **(not exist)**
            /home/surendrab/MDP/old_v_distro/lj-os/lib/:   (not exist)
            /home/surendrab/MDP/old_v_distro/lj-os/usr/lib/riscv64-unknown-linux-gnu/7.3.1/:  **(not exist)**
            /home/surendrab/MDP/old_v_distro/lj-os/usr/lib/   **(not exist)**

Here  /home/surendrab/MDP/old_v_distro/lj-os/     path is my host pc directory location.

Even I have tried -B -I and -L option:
# gcc -o out sample.c -I /usr/include/ -L /gcc-h/libexec/gcc/riscv64-unknown-linux-gnu/7.3.1 -B /gcc-h/bin/
**gcc: error trying to exec 'cc1': execvp: No such file or directory**

Here cc1 is there in /gcc-h/libexec/gcc/riscv64-unknown-linux-gnu/7.3.1 but when I am running gcc its giving 
**gcc: error trying to exec 'cc1': execvp: No such file or directory**.

Can you please suggest me what to do now ?

cmuellner, 2021-04-27:
If I understand correctly, you want to build a native RISC-V toolchain on your x86_64 PC, right?

So the general steps are:
* build cross compiler (incl. bootstrapping) with sysroot
* build native compiler for foreign architecture using the cross compiler

You did this using `make linux-native`, but you found out, that `/opt/riscv-native/native` does include the native compiler, but is missing the `sysroot` directory.

So you see something like this:
```
/opt/riscv-native$ ls
bin  include  lib  lib64  libexec  native  riscv64-unknown-linux-gnu  share  sysroot
/opt/riscv-native/native$ cd native
/opt/riscv-native/native$ ls
bin  include  lib  lib32  lib64  libexec  riscv64-unknown-linux-gnu  share
```

As Jim suggested, let's have a look at the paths of the (foreign) native compiler to understand the compiler better
(here executed on the host, but this could also be done on the target without invoking ld.so manually):

```
$ /opt/riscv-native/sysroot/lib/ld-linux-riscv64-lp64d.so.1 --library-path /opt/riscv-native/sysroot/lib64/lp64d ./bin/gcc --print-search-dirs
install: /opt/riscv-native/native/lib/gcc/riscv64-unknown-linux-gnu/10.2.0/
programs: =/opt/riscv-native/native/libexec/gcc/riscv64-unknown-linux-gnu/10.2.0/:/opt/riscv-native/native/libexec/gcc/riscv64-unknown-linux-gnu/10.2.0/:/opt/riscv-native/native/libexec/gcc/riscv64-unknown-linux-gnu/:/opt/riscv-native/native/lib/gcc/riscv64-unknown-linux-gnu/10.2.0/:/opt/riscv-native/native/lib/gcc/riscv64-unknown-linux-gnu/:/opt/riscv-native/native/lib/gcc/riscv64-unknown-linux-gnu/10.2.0/../../../../riscv64-unknown-linux-gnu/bin/riscv64-unknown-linux-gnu/10.2.0/:/opt/riscv-native/native/lib/gcc/riscv64-unknown-linux-gnu/10.2.0/../../../../riscv64-unknown-linux-gnu/bin/
libraries: =/opt/riscv-native/native/lib/gcc/riscv64-unknown-linux-gnu/10.2.0/lib64/lp64d/:/opt/riscv-native/native/lib/gcc/riscv64-unknown-linux-gnu/10.2.0/../../../../riscv64-unknown-linux-gnu/lib/riscv64-unknown-linux-gnu/10.2.0/lib64/lp64d/:/opt/riscv-native/native/lib/gcc/riscv64-unknown-linux-gnu/10.2.0/../../../../riscv64-unknown-linux-gnu/lib/../lib64/lp64d/:/lib64/lp64d/riscv64-unknown-linux-gnu/10.2.0/lib64/lp64d/:/lib64/lp64d/../lib64/lp64d/:/usr/lib64/lp64d/riscv64-unknown-linux-gnu/10.2.0/lib64/lp64d/:/usr/lib64/lp64d/../lib64/lp64d/:/lib/riscv64-unknown-linux-gnu/10.2.0/lib64/lp64d/:/lib/../lib64/lp64d/:/usr/lib/riscv64-unknown-linux-gnu/10.2.0/lib64/lp64d/:/usr/lib/../lib64/lp64d/:/opt/riscv-native/native/lib/gcc/riscv64-unknown-linux-gnu/10.2.0/:/opt/riscv-native/native/lib/gcc/riscv64-unknown-linux-gnu/10.2.0/../../../../riscv64-unknown-linux-gnu/lib/riscv64-unknown-linux-gnu/10.2.0/:/opt/riscv-native/native/lib/gcc/riscv64-unknown-linux-gnu/10.2.0/../../../../riscv64-unknown-linux-gnu/lib/:/lib64/lp64d/riscv64-unknown-linux-gnu/10.2.0/:/lib64/lp64d/:/usr/lib64/lp64d/riscv64-unknown-linux-gnu/10.2.0/:/usr/lib64/lp64d/:/lib/riscv64-unknown-linux-gnu/10.2.0/:/lib/:/usr/lib/riscv64-unknown-linux-gnu/10.2.0/:/usr/lib/
```

As you can see, for example in the last item `/usr/lib/`, the foreign native compiler uses `/` as sysroot.
I.e. it is meant to be used in combination with you RISC-V Linux distribution.
Be aware that this is already in the danger zone, as the libraries of your RISC-V Linux distribution might be incompatible with your toolchain (e.g. in case of an ABI change).

In case you want a foreign native compiler with a sysroot, then you have to patch Makefile.in as follows:

* Define NATIVE_SYSROOT=/$(INSTALL_DIR)/native/sysroot
* Adjust `stamps/build-gcc-linux-native` to become more similar to `stamps/build-gcc-linux-stage2`. So, setting the corresponding sysroot (maybe even a build-sysroot) as configure argument, and copying the sysroot.
* Do the previous step for all Makefile rules that copy something to $SYSROOT (e.g. install all built packages additionally to $NATIVE_SYSROOT).

jim-wilson, 2021-04-27:
> You did this using `make linux-native`, but you found out, that `/opt/riscv-native/native` does include the native compiler, but is missing the `sysroot` directory.

He didn't use riscv-gnu-toolchain.  He used some scripts I sent in another issue which use --with-sysroot= configure options.  These are mentioned at the top.  I try to strongly discourage people from using riscv-gnu-toolchain to build linux native compilers because it is very unlikiely to do anything useful.  You need a glibc of the same version with same patches and same configure options, which riscv-gnu-toolchain can't give you.

jim-wilson, 2021-04-27:
> Sorry I don't know what is mean by canadian cross built compiler

Canadian cross technically means build!=host!=target.  E.g using a linux build system for a windows hosted riscv64-elf target compiler.  But I sometimes use it for build!=host==target builds which is what you did.

> /home/surendrab/MDP/old_v_distro/lj-os/lib64/lp64d/: **(not exist)**

So on the linux build machine, the --with-sysroot= option pointed at /home/surendrab/MDP/old_v_distro/lj-os.  On your riscv64-linux system, you should create the /home/surendrab/MDP/old_v_distro dirs, and then create a symlink there that makes lj-os point at /, i.e. the root of the file system.  Then this path will work.

> Here cc1 is there in /gcc-h/libexec/gcc/riscv64-unknown-linux-gnu/7.3.1 but when I am running gcc its giving

So add a -B/gcc-h/libexec/gcc/riscv64-unknown-linux-gnu/7.3.1/ option so that gcc can find the cc1 binary.

These are temporary solutions.  When you get this cross built compiler working, you should use it to build gcc again natively, and this gcc will have all of the right paths built in and won't need any workarounds.

cmuellner, 2021-04-27:
> He didn't use riscv-gnu-toolchain. He used some scripts I sent in another issue which use --with-sysroot= configure options. These are mentioned at the top. I try to strongly discourage people from [...]

I fully agree that a foreign native GCC (without its own sysroot) will most likely not do anything useful.
The gcc binary might not even start on the target machine (because the glibc ABI is not compatible and ld.so bails out).

Having a foreign native toolchain incl. a sysroot with statically linked GCC binaries can help in some use-cases. With that, you can build static binaries of your code (on the target) and have a way to execute that in combination with your preferred combination of GCC/glibc/$yourlib. So you are more or less host-system agnostic (you only see the kernel interface). However, that needs quite some experience to understand what things you can do and cannot do/expect with such a toolchain.

billasurendra, 2021-04-28:
> Canadian cross technically means build!=host!=target. E.g using a linux build system for a windows hosted riscv64-elf target compiler. But I sometimes use it for build!=host==target builds which is [...]

Thank you jim. I have followed exactly what you have suggested. Now native compiler working on RISC-V Linux, I am able to compile sample C program. Thank you once again....

jim-wilson, 2021-05-04:
No, you just build the toolchain normally.

mkdir X-bin-build; cd X-bin-build
../binutils-gdb/configure --prefix=$installdir
make -j4
make install
mkdir X-gcc-build; cd X-gcc-build
../gcc/configure --disable-multilib --prefix=$installdir
make -j4
make install

$installdir can be any directory you have write access to.

billasurendra, 2021-05-05:
Hi Jim,

I have tried what you said.

> No, you just build the toolchain normally.
> 
> mkdir X-bin-build; cd X-bin-build
> ../binutils-gdb/configure --prefix=$installdir
This command I can successfully run and generated Make file also. But When I am trying to run make -j4 its giving error.

**Error:
-/bin/sh: make: not found**

I think make is not available in riscv target image. Can you please tell me what I have to install in target image for working make command on riscv target.

> make -j4
> make install
> mkdir X-gcc-build; cd X-gcc-build
> ../gcc/configure --disable-multilib --prefix=$installdir
> make -j4
> make install
> 
> $installdir can be any directory you have write access to.

billasurendra, 2021-05-05:
Downloaded a make-3.8-tar.gz source and cross compiled on host with riscv cross compiler, copied make_install folder to riscv target image. now again I have tried to execute **make -j4** but its giving different error.
 **Error mesage:**
**/X-binutils-build # make -j4**
make[1270]: unhandled signal 11 code 0x2 at 0x0000003fffea6ce8
CPU: 0 PID: 1270 Comm: make Not tainted 5.4.1 #1
sepc: 0000003fffea6ce8 ra : 0000003fca361e34 sp : 0000003fffea4940
 gp : 00000000000387f8 tp : 0000003fca4081e0 t0 : 0000000000000111
 t1 : 000000000000005d t2 : 00000000000382e0 s0 : 0000003fffea4ca0
 s1 : 0000003fca3ec4e8 a0 : 0000003fffea4940 a1 : 0000003fffea49a0
 a2 : 0000000000000005 a3 : ffffffc0001adc7e a4 : 0000000000000000
 a5 : 0000003fffea6ce8 a6 : 00000000000525c0 a7 : 000000000000002a
 s2 : 0000000000000001 s3 : 0000003fffea4940 s4 : 00000000000525c0
 s5 : 0000000000000005 s6 : 0000000000000001 s7 : 0000000000000204
 s8 : 0000000000001000 s9 : 0000000000000000 s10: 0000003fffea51f8
 s11: 0000000000000217 t3 : 8101010101010100 t4 : 0000000000000002
 t5 : 0000003fca2bee70 t6 : 0000000000000000
sstatus: 8000000000006020 sbadaddr: 0000003fffea6ce8 scause: 000000000000000c
Segmentation fault

Can you please tell me, what might be the reason for above error.
Thanks
Billa

jim-wilson, 2021-05-05:
If you don't have make or other common utillities on your RISC-V target then you will have to cross compile them.

I have no idea why your make failed.  I would suggest that you copy the core file from the RISC-V target to the build machine, and then run a cross gdb on the binary and core file to see where it failed.

billasurendra, 2021-06-02:
Dear Jim,

With help of you, now I am having native compiler in my riscv target machine with tempory path setup. Now I am planning to compile binutils and gcc with this compiler. I have started compiling binutils first. when I am running ./configure command  in binutils directory its showing to some packages no. Here I am sharing printed statements when I am running ./configure command, I am struggling to enable five package in riscv target image. Can you please help me out how I can cross compile and enable them on target riscv. Please look into highlighted statements and give your comments. 

$ ./configure

checking build system type... riscv64-unknown-linux-gnu
checking host system type... riscv64-unknown-linux-gnu
checking target system type... riscv64-unknown-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking for gawk... gawk
checking for gcc... gcc
checking for C compiler default output file name... a.out
checking whether the C compiler works... yes
checking whether we are cross compiling... no
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking whether g++ accepts -static-libstdc++ -static-libgcc... yes
**checking for gnatbind... no**
**checking for gnatmake... no**
**checking whether compiler driver understands Ada... no**
checking how to compare bootstrapped objects... cmp $$f1 $$f2 16 16 
checking for objdir... .libs                                       
**checking for isl 0.15 or later... no**                  
required isl version is 0.15 or later                               
checking for default BUILD_CONFIG...                               
checking for --enable-vtable-verify... no               
checking for bison... bison -y                                      
checking for bison... bison                                        
checking for gm4... no                                  
checking for gnum4... no                                            
checking for m4... m4                                              
checking for flex... flex                               
checking for flex... flex                                           
checking for makeinfo... makeinfo                                  
checking for expect... no            
checking for runtest... runtest                                      
checking for ar... ar                                              
checking for as... as                                               
checking for dlltool... no                            
ecking for windmc... no                                          
checking for objcopy... objcopy                         
checking for objdump... objdump                                     
checking for readelf... readelf                                    
checking for cc... no                                 
checking for gcc... gcc                                             
checking for c++... c++                                            
checking for gcc... gcc                                 
**checking for gfortran... no**                                        
**checking for gccgo... no**                                         
checking whether to enable maintainer-specific portions of Makefiles... no
configure: creating ./config.status                       
config.status: creating Makefile

jim-wilson, 2021-06-02:
You only need to worry about configure messages if configure exits with an error and refuses to create a Makefile.  That didn't happen here.

Ada and Go are non-default languages in GCC.  You build them only if you need them.  The toplevel configure just checks for them so it knows what libraries can be built.  Since the Ada compiler is written in Ada, if you want Ada, you need to start with a cross compiler same as you already did.  I think with Go you can just built it natively as it is written in C.

Fortran is a default language for GCC.  You must have configured without it.  But it is written in C, so you can just build it natively using the compiler you already have.  Again, it is checked for here only so configure knows what libraries can be built.

isl is an optional library for high level loop optimizations.

All of this (except Ada) is stuff you can build later if you need it.  You just need to get a native compiler built first, and then you can start building applications and libraries, and then later, if you want, you can build another native compiler with more features.

billasurendra, 2021-06-03:
./configure command generating Makefile without these (ada, fortran, go, isl). But when I am started running make -j8 command it is printing first 4 lines different after that printing configuration time statements (like this way it is repeatedly printing, its not stoping). below I am sharing first 4 line statements.  is there any solution for successful running of make -j8. 

/binutils-2.30 # make -j8
make: Warning: File 'Makefile.in' has modification time 1517065560 s in the future
CONFIG_SHELL="/bin/sh" /bin/sh ./config.status --recheck
running CONFIG_SHELL=/bin/sh /bin/sh ./configure --prefix=/X-bin --no-create --no-recursion
checking build system type... riscv64-unknown-linux-gnu

jim-wilson, 2021-06-03:
If you are NFS mounting a disk, then the time on the server and client are out of sync.  If there is only one system involved, then maybe the time on this machine is unstable.  Enable a network time service to get a reliable time.  Or maybe the system always has the same wrong time, and you untarred files that were tarred up on another system with the correct time.  Anyways, the solution is all the same.  You need a correct date and time on the system.  And check the date and time on the files.  If they are wrong, then you will need to fix them.

billasurendra, 2021-06-04:
Fantastic answer Jim. It's working now. make -j8 command working now.

Thanks for your help
Billa

billasurendra, 2021-07-07:
Hi Jim,
Hi,

I am trying to compiling **libbffi** with native compiler for RISC-V architecture but I am getting the following error. without libffi I am not able to cross compile python3.

Error: 

configure: error: "libffi has not been ported to riscv64-unknown-linux-gnu."

Can you please help me regarding this.

Thanks
Billa

kito-cheng, 2021-07-07:
I saw upstream libffi already support riscv? which version you are using?

https://github.com/libffi/libffi/tree/master/src/riscv

billasurendra, 2021-07-07:
libffi-3.1

kito-cheng, 2021-07-07:
RISC-V port are added after libffi 3.3

billasurendra, 2021-07-07:
Even I have tried this **https://github.com/libffi/libffi/tree/master/src/riscv** . Here when I am running ./autogen, I am getting error.

Error:
configure:7901: error: possibly undefined macro: AC_PROG_LD
autoreconf: /usr/local/bin/autoconf failed with exit status: 1

billasurendra, 2021-07-07:
Thanks libffi 3.4.2 worked for me now.

billasurendra, 2021-07-11:
Hi Jim,
I am trying to generate a riscv native compiler with your procedure, but this time I am using another version compiler than previous i.e gcc-7.3.1. Here I have properly copied target file system files for sysroot. Started compiling 4 scripts suggested by you.
1. x-binutils:  was sucessfull

../binutils-2.30/configure --prefix=/home/surendrab/MDP/old_v_distro/sources/x-bin --target=riscv64-unknown-linux-gnu --with-sysroot=${LJOS} --disable-multilib

2. x-gcc: is unsuccessful, its giving error.

../gcc-7.3.1/configure --target=riscv64-unknown-linux-gnu --prefix=/home/surendrab/MDP/old_v_distro/sources/x-gcc --with-sysroot=${LJOS} --disable-multilib --with-arch=rv64imafdc --enable-languages=c,c++ -with-gmp=/home/surendrab/MDP/old_v_distro/sources/gmp-6.1.2 -with-mpfr=/home/surendrab/MDP/old_v_distro/sources/mpfr-4.0.1 -with-mpc=/home/surendrab/MDP/old_v_distro/sources/mpc-1.1.0

Error Details are here:

checking for suffix of object files... configure: error: in `/home/surendrab/MDP/old_v_distro/native_compiler/x-gcc/riscv64-unknown-linux-gnu/libgcc':
configure: error: cannot compute suffix of object files: cannot compile
See `config.log' for more details.
make[1]: *** [Makefile:13085: configure-target-libgcc] Error 1
make[1]: Leaving directory '/home/surendrab/MDP/old_v_distro/native_compiler/x-gcc'
make: *** [Makefile:899: all] Error 2

Can you please help me out regarding this error. I am struggling to resolve this error.

Thanks in advance
Billa

jim-wilson, 2021-07-11:
You have to look at the config.log file in the directory where the build failed, find the command that failed, and then figure out why it failed.  So it looks like riscv64-unknown-linux-gnu/libgcc/config.log is the file you need to look at.

billasurendra, 2021-07-11:
It has exited with error exit:1. I am not getting any clue with this config
file.I am sharing config.log here please check once.

billasurendra, 2021-07-11:
I have looked at confog.log file. I have found my command like this.

  $
/home/surendrab/MDP/old_v_distro/native_compiler/gcc-7.3.1/libgcc/configure
--srcdir=../../../gcc-7.3.1/libgcc --cache-file=./config.cache
--with-cross-host=x86_64-pc-linux-gnu
--prefix=/home/surendrab/MDP/old_v_distro/native_compiler/x-gcc
--with-sysroot=/home/surendrab/MDP/old_v_distro/lj-os --disable-multilib
--enable-languages=c,c++,lto
*--program-transform-name=s&^&riscv64-unknown-linux-gnu-&
--disable-option-checking* --with-target-subdir=riscv64-unknown-linux-gnu
--build=x86_64-pc-linux-gnu --host=riscv64-unknown-linux-gnu
--target=riscv64-unknown-linux-gnu

In this command highlighted one is not given by me. Is it creating a
problem ?.

jim-wilson, 2021-07-20:
The config.log file should have info about a series of tests that configure is performing to figure out what features are available.  Some tests are optional, e.g. if there is more than one way to do something, we only need one of the checks to succeed.  Some tests are required.  E.g. if the compiler doesn't work for a trivial program then we can't do anything at all.  The test that failed in your case is a required test to verify that the compiler works.  So you need to search through the config.log file to find that particular test, and then figure out why that test failed.  The error message gave you useful info like "checking for suffix of object files" so you should look for that string or a similar one in the config.log file to find the test that failed.

TommyMurphyTM1234, 2022-09-27:
Multiple questions about RISC-V GCC toolchain and other library build issues but nothing that is clearly an issue with the RISC-V GCC toolchain.

closed 2022-09-27 by TommyMurphyTM1234