riscv-collab/riscv-gnu-toolchain #883: GCC for RVV intrinsic produces redundent instructions

hawajkm, 2021-04-16:
Hello, everyone!

I was debugging an issue with the [rvv-intrinsic](https://github.com/riscv/riscv-gnu-toolchain/tree/rvv-intrinsic) branch of the toolchain where random `vmv1r` would appear with the source and the destination being the same. In my quest, I turned to inline assembly to further understand the issue. So, I made this example

```
int input0[]          = { 0, 1, 2, 3, 4, 5, 6, 7 };                             
volatile int output[] = { 0, 0, 0, 0, 0, 0, 0, 0 };                             
int main(void) {                                                                
  int vlen = vsetvl_i(8);                                                       
  vint32m1_t a;                                                                 
  asm ( \                                                                       
    "vle32.v %[val], 0(%[base]) \n"                                             
    : [val] "=vd" (a)                                                           
    : [base] "r" (input0)                                                       
    : "memory"                                                                  
  );                                                                            
  vint32m1_t b = vadd_vv_i32m1(a, a);                                           
  vse32_v_i32m1((int*)output, b);                                               
  return 0;                                                                     
}     
```

If I compile that as follows:
```
$ riscv64-unknown-linux-gnu-g++ -O3 -static example.cc -o example
```

Then, I disassembled the output to see what the program compiles to, as follows:
```
$ riscv64-unknown-linux-gnu-objdump -D example > example.asm
```

Here is what I observe:
```
0000000000010504 <main>:                                                        
   10504: db818793            addi  a5,gp,-584 # 6cf80 <output>                 
   10508: 0006b737            lui a4,0x6b                                       
   1050c: 01007057            vsetvli zero,zero,e32,m1,tu,mu,d1                 
   10510: 00070713            mv  a4,a4                                         
   10514: 4501                  li  a0,0                                        
   10516: 02076087            vle32.v v1,(a4)                                   
   1051a: 9e1030d7            vmv1r.v v1,v1                                     
   1051e: 021080d7            vadd.vv v1,v1,v1                                  
   10522: 0207e0a7            vse32.v v1,(a5)                                   
   10526: 8082                  ret
```

I don't really understand why the `vmv1r.v` is required there; especially if the destination and source are the same. Is there anything I am missing or is this an issue with the toolchain?

I tried looking through the definitions of how moves [are defined](https://github.com/riscv/riscv-gcc/blob/66a2be891b672cdb3088d44a7c50939675f8302d/gcc/config/riscv/vector.md#L966-L978) for vector, and I am confused why the `mode` is set to `none` in here. I am not sure if it is related to the issue on hand, but I presume mimicking the scalar should be the correct way.

```
(define_insn "*mov<mode>"
  [(set (match_operand:VMODES 0 "reg_or_mem_operand"  "=vr,vr,m")
	(unspec:VMODES
	  [(match_operand:VMODES 1 "reg_or_mem_operand"  "vr,m,vr")
	   (reg:SI VL_REGNUM)]
	 UNSPEC_USEVL))]
  "TARGET_VECTOR"
  "@
   vmv<lmul>r.v\t%0,%1
   vl<lmul>r.v\t%0,%1
   vs<lmul>r.v\t%1,%0"
  [(set_attr "type" "vector")
   (set_attr "mode" "none")])
```

Thanks!

qtt2, 2021-04-20:
Hi Kito Cheng (@kito-cheng) and Andrew Waterman (@aswaterman),

Do you have any pointer to where we should look into to address the issue above?

Thanks!

jim-wilson, 2021-04-20:
You can use -fdump-rtl-all to see the output of each RTL optimization pass, and look for a place inserting the extra instruction, or failing to remove it.  But note that the gcc vector support is not well optimized, and not being actively worked on.  You should not expect high performance from it.  Current focus is on LLVM vector support.

There is no reasonable setting for the mode attribute, since this is clearly not a SImode or DImode etc instruction.  But the modes  are only used for identifying double word move instructions which expand into two assembly insns.  This mode attribute is useless for the vast majority of instruction patterns in the md files.  Thus it is harmless to define it away.

hawajkm, 2021-04-20:
Thank you very much! I will take at the output. Hopefully, I can use gcc as a stop-gap solution for the time being.

I will be looking forward for the LLVM port. But, I do need something for the near future, so let me see if I can remedy gcc to do what I need.

hawajkm, 2021-04-20:
Hi Jim Wilson (@jim-wilson),

Okay, we have been playing around with the machine description file for vector. I don't think we made any meaningful progress. However, we did notice one more aspect to the issue that escaped us; if you look over the small example we have posted, you can see that there are redundant move instructions generated for _scalar_ registers as well:
```
   1050c: 01007057            vsetvli zero,zero,e32,m1,tu,mu,d1                 
 >>10510: 00070713            mv  a4,a4                                         
   10514: 4501                li  a0,0                                        
```

Now, some observations we have made about the implementation of the `define_insn` in the machine description (MD) file for vector:

1. The `type` for _all_ vector instructions has been defined as `vector`; while for _scalar_, a more elaborate and complex typing has been defined. This pushed us to think that _maybe_ the typing is used in the scalar part to perform optimizations? Moreover, is the fact that all vector instructions have the same `type` an initial step and the plan is to implement a more complex and complete typing eventually?
2. Most vector instructions have been tagged/flagged? with `UNSPEC_USEVL`. Although we don't know what this flag/tag does, it seems that it has to do with vector length specification?
3. Is the specification for vector instructions complete, but they are missing optimization passes? Or are the vector instructions not implemented/specified completely and some flags are required to leverage native gcc passes?
4. As you can tell, we are pretty novice when it comes to GCC development. We have a simple requirement to use vector intrinsics in GCC. How much effort, in your opinion, is required to get a decently optimized version of GCC (reads: doesn't have to be perfect nor the best)?

Thank you very very much for have the patience to help us navigate this! :)

jim-wilson, 2021-04-20:
The type attributes are primarily used for instruction scheduling.  There is no vector instruction scheduling support as yet.

The vector instruction support should be complete, but there is no autovectorization, and no vector specific optimization passes.  There is no attempt to optimize anything at all, other than to try to remove some redundant vl sets.  Since this is not actively maintained, it may not be fully up to date with the evolving vector ISA spec and vector programming model.

Since the vector length can be changed dynamically, we need to set up dependencies between an instruction that uses a vl and the instruction that sets the vl.  This is what the unspec_vl is used for.  The intrinsics actually emit two instructions, one to set vl and one that uses it for the vector operation, and then we try to remove the redundant set vl instructions, since typically vl does not change very often.  We need dependencies between the instructions for this to work, and we need to be able to distinguish different vl values.

A proper vector compiler is many man years of work.  But some of this work can be done incrementally.  It is hard to say how much is required to get what you want.

Register allocation is a hard problem.  It is hard to say anything useful about what is going on without someone spending time looking at it.

lazyparser, 2021-04-21:
> A proper vector compiler is many man years of work. But some of this work can be done incrementally. It is hard to say how much is required to get what you want.

Indeed. Although the PLCT Lab is willing to improve the codegen/opt for GCC in GCC, it is simply too hard for our  young engineers now. It would be appreciate if Kito or Jim provides a few guidance/hints/roadmaps on how to achieve the goal (on par with AVX/SVE).

TommyMurphyTM1234, 2022-09-28:
Would it be possible to review this issue in the light of the latest and planned work in the `rvv-next` branch (https://github.com/riscv-collab/riscv-gnu-toolchain/tree/rvv-next) to see what issues are still extant and what, if anything, can be done about them?

TommyMurphyTM1234, 2022-10-10:
I tried to check the original issue with the latest tools built as follows:
```
git clone https://github.com/riscv/riscv-gnu-toolchain rv64-linux-rvv-next -b rvv-next
cd rv64-linux-rvv-next
./configure --prefix=`pwd`/installed-tools
date > build.log; make linux 2>&1 | tee -a build.log; date >> build.log
```
But I got this - presumably because of changes to the vector extension standard/implementation since the original issue was logged?
```
tommy@pelican:~/Downloads/riscv-tools/rv64-linux-rvv-next/installed-tools/bin$ ./!!
./riscv64-unknown-linux-gnu-g++ -O3 -static example.cc -o example
example.cc:6:9: warning: backslash and newline separated by space
    6 |   asm ( \
      |          
example.cc: In function 'int main()':
example.cc:4:14: error: 'vsetvl_i' was not declared in this scope
    4 |   int vlen = vsetvl_i(8);
      |              ^~~~~~~~
example.cc:5:3: error: 'vint32m1_t' was not declared in this scope
    5 |   vint32m1_t a;
      |   ^~~~~~~~~~
example.cc:8:20: error: 'a' was not declared in this scope
    8 |     : [val] "=vd" (a)
      |                    ^
example.cc:12:13: error: expected ';' before 'b'
   12 |   vint32m1_t b = vadd_vv_i32m1(a, a);
      |             ^~
      |             ;
example.cc:13:31: error: 'b' was not declared in this scope
   13 |   vse32_v_i32m1((int*)output, b);
      |                               ^
example.cc:13:3: error: 'vse32_v_i32m1' was not declared in this scope
   13 |   vse32_v_i32m1((int*)output, b);
      |   ^~~~~~~~~~~~~
```
Does anybody with better knowledge than I have (i.e. basically none! :-) of the vector extensions/implementation have any ideas on how to do an appropriate test here please?

TommyMurphyTM1234, 2022-10-10:
In order to investigate this I build the latest `rvv-next` tools as follows:
```
git clone https://github.com/riscv/riscv-gnu-toolchain rv64-linux-rvv-next -b rvv-next
cd rv64-linux-rvv-next
./configure --prefix=`pwd`/installed-tools --with-arch=rv64gcv
date > build.log; make linux 2>&1 | tee -a build.log; date >> build.log
```
The original code did not work:
```
int input0[]          = { 0, 1, 2, 3, 4, 5, 6, 7 };                             
volatile int output[] = { 0, 0, 0, 0, 0, 0, 0, 0 };                             
int main(void) {                                                                
  int vlen = vsetvl_i(8);                                                       
  vint32m1_t a;                                                                 
  asm ( \                                                                       
    "vle32.v %[val], 0(%[base]) \n"                                             
    : [val] "=vd" (a)                                                           
    : [base] "r" (input0)                                                       
    : "memory"                                                                  
  );                                                                            
  vint32m1_t b = vadd_vv_i32m1(a, a);                                           
  vse32_v_i32m1((int*)output, b);                                               
  return 0;                                                                     
}   
```
I had to change it as follows to get it to compile without errors/warnings:
```
#include <riscv_vector.h>

int input0[]          = { 0, 1, 2, 3, 4, 5, 6, 7 };                             
volatile int output[] = { 0, 0, 0, 0, 0, 0, 0, 0 };                             
int main(void) {                                                                
  int vlen = vsetvl_e8m8(8);                                                       
  vint32m1_t a;                                                                 
  asm (                                                                   
    "vle32.v %[val], 0(%[base]) \n"                                             
    : [val] "=vd" (a)                                                           
    : [base] "r" (input0)                                                       
    : "memory"                                                                  
  );                                                                            
  vint32m1_t b = vadd_vv_i32m1(a, a, 8); 
  vse32_v_i32m1((int*)output, b, 8);                                               
  return 0;                                                                     
}
```
I am not sure if the length of `8` passed to a few functions is correct?

The disassembly of `main()` now looks like this:
```
0000000000010556 <main>:
   10556:	00071737          	lui	a4,0x71
   1055a:	00070713          	mv	a4,a4
   1055e:	dc818793          	addi	a5,gp,-568 # 72eb0 <output>
   10562:	02076c07          	vle32.v	v24,(a4)
   10566:	c5047057          	vsetivli	zero,8,e32,m1,ta,mu
   1056a:	4501                	li	a0,0
   1056c:	038c0c57          	vadd.vv	v24,v24,v24
   10570:	0207ec27          	vse32.v	v24,(a5)
   10574:	8082                	ret
```
@kito-cheng - (a) Are my changes to the text code OK and (b) does this show that the original issue is no longer extant?
Thanks

TommyMurphyTM1234, 2022-10-23:
Gentle reminder ... would anybody, in particular @zhongjuzhe or @kito-cheng, be able to comment on my previous post and clarify if this issue can be closed at this stage, please? Thanks.

zhongjuzhe, 2022-10-23:
Yes，close it---- Replied Message ----FromTommy ***@***.***>Date10/23/2022 19:15 ***@***.***> CcJu-Zhe ***@***.******@***.***>SubjectRe: [riscv-collab/riscv-gnu-toolchain] GCC for RVV intrinsic produces redundent instructions (#883)
Gentle reminder ... would anybody, in particular @zhongjuzhe or @kito-cheng, be able to comment on my previous post and clarify if this issue can be closed at this stage, please? Thanks.

[
{
"target": "https://github.com/riscv-collab/riscv-gnu-toolchain/issues/883#issuecomment-1288088900",
"url": "https://github.com/riscv-collab/riscv-gnu-toolchain/issues/883#issuecomment-1288088900",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

TommyMurphyTM1234, 2022-10-23:
Thanks @zhongjuzhe.

closed 2022-10-23 by TommyMurphyTM1234