riscv-collab/riscv-gnu-toolchain #884: No g++ find in bin/

Mophe, 2021-04-18:
hello! 

I am confused with a problem. After I enter "make linux", i can't find g++ in opt/riscv/bin. No error occured after "make". How to solve the problem? Thank you!

jim-wilson, 2021-04-18:
Is there anything in /opt/riscv/bin?  Does make linux report anything?

If you are on a recent linux distro, you can just apt/dnf/yum/whatever install a riscv64-linux compiler.  It won't be multilib, but it will work fine for rv64gc/lp64d code.

Mophe, 2021-04-19:
Thanks for your help!

My experiment requires `g++` with other instruction sets like `RV64IC` & `RV64I` ， so I am troubled by this problem.

No errors were reported at the end. I'm not sure if there was an error in the middle. `/opt/riscv/bin` is not null, but `riscv64-unknown-linux-gnu-g++` not in there.

I'll list files in opt/riscv/bin.(Both `make` and `make linux` have been used)

`Mo@node:/opt/riscv/bin$ ls riscv64-unknown-elf-*`
[image]

`Mo@node:/opt/riscv/bin$ ls riscv64-unknown-linux-gnu-*`
[image]

jim-wilson, 2021-04-19:
The build process does binutils, then stage1 gcc, then the C library, then stage2 gcc.  Stage1 gcc only includes the C compiler.  So the build either failed in the library build or in stage2 gcc build.  You can tell the different if you check to see if the C library is there.  Try using gcc to compile a hello world.  If that fails then the C library wasn't built.  If hello world works, then the C library was built.

To find the build error, save the build result in a file and then look for the string "error".  E.g.
  make &> build.log
and then look for an error.  The compiler will print "error:"  Make will print error with some asterisks *.  If you know which part failed to build, you can cd into that directory and run make to try to rebuild it.  If you type make at the top level, it will rm -rf the failed build dir and then reconfigure and rebuild which takes longer to get the error again.

Mophe, 2021-04-19:
Thanks a lot!

After I save the build result in a file, I find an error occurred in `make linux`.

```
In file included from /home/Mo/riscv-gnu-toolchain/build/../riscv-gdb/gdb/python/py-tui.c:32:0:
/home/Mo/riscv-gnu-toolchain/build/../riscv-gdb/gdb/gdb_curses.h:57:18: fatal error: term.h: No such file or directory
compilation terminated.
```
then I find a similar questions in [stackoverflow](https://stackoverflow.com/questions/25920535/term-h-header-not-found), and I install `libncurses5-dev`, In the next `make linux`, the error no longer appears and g++ is functioning well.

thansk again!

TommyMurphyTM1234, 2022-09-27:
Original issue seems to have been with the user's build environment and that has been resolved so closing this issue now.

closed 2022-09-27 by TommyMurphyTM1234