riscv-collab/riscv-gnu-toolchain #885: GROMACS on RISC-V

amithmath, 2021-04-19:
I was wondering, is there any possibilities to run GROMACS on RISC-V (32 bit)? As far as I know, it requires vector extensions like mavx2 and mfma on x86. I am not whether there exist similar extension  for RISC-V.

amithmath, 2021-04-19:
When I try to compile GROMACS I am getting following errors-

riscv32-unknown-elf-c++ error: unrecognized command line option '-mavx2'
riscv32-unknown-elf-c++: error: unrecognized command line option '-mfma'

jim-wilson, 2021-04-19:
GROMACS is not part of the toolchain.  I would suggest asking the GROMACS community for help.  -mavx2 is an x86 compiler option.  Maybe it doesn't support cross compilation?  Or maybe you need special configure options?  The GROMACS community are the ones that would know.

RISC-V has the V (vector) and P (packed) extensions which are in draft, but it may be a few years before you can get hardware that supports them, and compilers have good support for them.  Qemu has support for one of the drafts of the V spec, I don't know about P.  But the compiler support will still be a problem.

amithmath, 2021-04-20:
I don't think it supports cross compilation.  I was going through pdf: https://people.eecs.berkeley.edu/~krste/papers/EECS-2016-1.pdf on page 75 there are various benchmarks, gromacs is one of them, it has been compared with RV32 and RV64 with other architectures. I was wondering, how they did it?

jim-wilson, 2021-04-20:
They probably cross compiled.  But this isn't a toolchain problem, it is a gromacs problem.  Try asking elsewhere, like the gromacs community, or the RISC-V sw-dev mailing list.

amithmath, 2021-04-20:
ok thanks

amithmath, 2021-04-20:
How do I cross compile GROMACS for RISC-V architecture?

amithmath, 2021-04-22:
While I was cross compiling gromacs, I am getting following error:

‘PTHREAD_MUTEX_INITIALIZER’ was not declared in this scope; did you mean ‘TMPI_THREAD_MUTEX_INITIALIZER’?

I think this error is connected to pthread.h I am not sure whether RV32 elf support pthread?

kito-cheng, 2021-04-22:
I guess you might need RV32 linux toolchain rather than RV32 elf toolchain?

amithmath, 2021-04-22:
I am not sure whether to use RV32 linux tool chain...but wondering why RV32 elf won't support pthread?

kito-cheng, 2021-04-22:
riscv32-unknown-elf toolchain is used for bare-metal environment, which means there might no OS there, that means no one to handle something like content switch or process scheduling, however you can goolge "bare metal thread" to find few thread implementation on bare-metal environment, but that's very target dependent.

But I guess that's not is what you interested at this moment, GROMACS seems like used on HPC, so I assume you might more interested run that on Linux or *BSD system.

amithmath, 2021-04-22:
OK got it...

closed 2021-04-22 by amithmath