riscv-collab/riscv-gnu-toolchain #887: Cpp Compiling Problem

kian-design, 2021-04-22:
Hi... I have found a code in cpp and it works pretty fine in visual studio 2019... I want to compile this code to RiscV with your toolchain but it throws some errors... I asked some programmers and they say it is due to diffrent standards of cpp and they say probably riscv gnu toolchain dont support it... which standards your compiler support in cpp? By the way i can send the error but it is very long...

kito-cheng, 2021-04-22:
You could check with GCC manual, and the language standard can be controlled by `-std=*` option.
https://gcc.gnu.org/onlinedocs/gcc-10.3.0/gcc/

Kian75, 2021-04-27:
Sorry but I searched a lot about the errors but I still have many of them left. I changed the version with -std to c++14 but I still have problems... I send you the errors if you have time please take a look.. the code is for cnn networks and I want to compile it to riscv.
[error.txt](https://github.com/riscv/riscv-gnu-toolchain/files/6383318/error.txt)

kito-cheng, 2021-04-27:
Seem like you need linux toolchain rather than elf toolchain.

Kian75, 2021-04-27:
I don't really know what is is the difference between these two :(( please don't judge me!.... what can I do to compile this code? In installation I installed both newlib and linux... if you say I need linux toolchain how can I try and compile this code with that?

cmuellner, 2021-04-27:
Looking at your error log, I cannot see a single RISC-V specific issue.

The Microsoft compiler and GCC are different in many aspects.
If your code compiles with the Microsoft compiler, but does not compile with GCC,
then you need to work on you code and not report bugs here.
Most of your issues might be covered on Stackoverflow.

I recommend, that you take a Linux distribution of your choice and work on getting the code compiling on that
(with the distribution's compiler). After that is done, you can have a look at cross compiling the code for RISC-V.

Two hints for compiling C++ code with GCC:
* use `g++` for compiling C++ code, not `gcc`
* Use ` -pthread` when using futures

Kian75, 2021-04-27:
Thank you for your help... I will try to solve the problem in the way that you mentioned....

TommyMurphyTM1234, 2022-09-27:
See @cmuellner's previous post.
Closing as invalid.

closed 2022-09-27 by TommyMurphyTM1234