riscv-collab/riscv-gnu-toolchain #889: How to compile toolchain for linux-musl?

alexthomas1, 2021-04-27:
Sorry if this is a really easy question but I am trying to figure this out but how do I compile the toolchain for musl? I tried using this flag in configure --with-musl-src, but that did not work

cmuellner, 2021-04-27:
What exactly did not work?

Here is how I would expect it to work:

```
./configure --prefix=/opt/riscv-musl --with-musl-src=/path/to/musl/
make -j$(nproc) musl
```

alexthomas1, 2021-04-27:
Thanks for the quick response! I see, so I need to provide my own musl source?

cmuellner, 2021-04-27:
Yes, you need to provide external source for musl.

The git submodules just include:
* binutils
* gcc
* glibc
* dejagnu
* newlib
* gdb
* qemu

cmuellner, 2021-04-27:
I just did a test build with the instructions above and can confirm that the toolchain builds with external musl sources (git `aad50fc`).

TommyMurphyTM1234, 2022-09-28:
@cmuellner - I think that this can be closed as completed given that `make musl` (to build a RISC-V Linux toolchain using musl instead of glibc) is now a valid make target which is documented in the README:

* https://github.com/riscv-collab/riscv-gnu-toolchain#installation-linux

cmuellner, 2022-09-29:
Yes, let's close this.

closed 2022-09-29 by cmuellner