riscv-collab/riscv-gnu-toolchain #890: Where to find Floating Point Number Representation?

AayushMisra, 2021-04-28:
Hie I was wondering at what stage the conversion of a number into IEEE 754 floating point representation takes place, and how to explore it further in the toolchain.

jim-wilson, 2021-04-28:
I don't think that I understand what question you are asking.

Numbers are always IEEE 754.  The only conversion is from the string in the source code to the binary IEEE 754 format.  See real_from_string in riscv-gcc/gcc/real.c.  GCC has a lot of complexity in FP handling because it supports cross compilation, so numbers are held in the target format, and we have a lot of code to do arithmetic on target format numbers without using the host FP instructions, just in case the host FP format is different from the target FP format.  This code is all in real.c and real.h.

AayushMisra, 2021-05-16:
Thanks Jim, your suggestions helped me a lot to get some insight on how real numbers are dealt in compilers, I have some follow up questions:
1. Which routine is responsible for choosing the appropriate real_format, and responsible for calling encode and decode functions of a specific real_format.
2. What is difference between 'decimal_single_format' and 'ieee_single_format' structure.
3. Unable to find definition for 'TARGET_FLOAT_FORMAT' and 'HOST_FLOAT_FORMAT' target/host description macros defining storage layout.
4. What changes are required to experiment with different formats, many different formats are already defined in real.c, how can i use them.

jim-wilson, 2021-05-19:
The encoding is in the internal language mode, so for instance SFmode is ieee_single_format as defined in machmode.def file.  And config/rs6000/rs6000-modes.def defines an IFmode that uses ibm_extended_format.  The C float type maps to SFmode.  C long double for the ppc target possibly maps to IFmode, or another mode, depending on compiler configuration and command line options.

ieee_single_format is IEEE binary FP format.  decimal_single_format is a decimal FP format.  I don't know the details, but probably something like every four bits specify a decimal digit.  decimal FP is popular with financial firms, which don't want inexact conversions to binary, and binary rounding.  They want numbers calculated exactly the same way humans calculate them.

TARGET_FLOAT_FORMAT was removed in 2008.  I think HOST_FLOAT_FORMAT was removed before 2000.  I think xm-machine.h files were removed in 1998.

To experiment with different float formats, you can change the language front end to map types to a different mode, or you can modify a target port to map a type to a different mode, possibly based on a command line option.  There are a number of examples to look at, for instance several targets have two different 128-bit long double encoding, with the choice depending on compiler configuration and/or command line options.  I suppose you could try hacking the machmode.def file to substitute a different encoding for standard mode like SFmode but that risks breaking the compiler because there are likely assumptions about SFmode scattered through the compiler that break if it isn't ieee_single_format.  Better is to fix the language front end or the target port to map types to a different mode.

AayushMisra, 2021-08-13:
Thanks for your guidance, indeed hacking the machmode.def file breaks the compiler multiple targets fail from various modules.
Can you elaborate on fixing the front end of C language, How can I create new datatypes and map them to different formats and what files would I need to manipulate?

jim-wilson, 2021-08-13:
When you say that you want to experiment with different formats, what exactly do you mean by this?  There are different changes you would need for different kinds of format changes.

if you want to try something that gcc already supports, that is easier.  For instance, we already have support for multiple 128-bit FP formats, so changing long double to use a different 128-bit FP format is pretty simple.  For instance, in gcc/config/rs6000/rs6000.c  we have
      if (TARGET_LONG_DOUBLE_128 && !TARGET_IEEEQUAD)
        REAL_MODE_FORMAT (TFmode) = &ibm_extended_format;
TFmode is the 128-bit long double format, which is ieee_quad_format by default, and the rs6000 (powerpc) port lets you choose to use either ieee or the ibm format.

If you want to define your own FP format, like a posit type, then you need a lot more changes.  You need to define your own mode that maps to that new FP type.  For instance, gcc/config/rs6000/rs6000-modes.def has
/* IBM 128-bit floating point.  */
FRACTIONAL_FLOAT_MODE (IF, FLOAT_PRECISION_IFmode, 16, ibm_extended_format);
to define IFtype.  If you do this, then you need patterns in the $target.md file for all operations on the new mode, which means specifying instructions that can operate on this mode/type.

You need changes to real.c and real.h to implement arithmetic on the FP type.

You may need changes to the C front end to add new keywords to you can have variable declarations that use the type.  That likely gets complicated fast, as you need to define how the new keywords interact with every C language feature.  And if you want C++ for Fortran then that is even more work.  But for C, I would start with c-family/c-common.c and search for "float" including the quotes, then try to make your new keyword work the same as float in every situation.

Easier would be to steal an existing keyword, e.g. make float map to your new FP type instead of ieee_single_float.  This can be done same as how rs6000 handles long double as mentioned at the top.

amithmath, 2021-10-10:
_>And_ if you want C++ for Fortran then that is even more work. But for C, I would start with c-family/c-common.c and search for >"float" including the quotes, then try to make your new keyword work the same as float in every situation._

Can you please tell if one want to add new keywords for FORTRAN and C++ which files to modify?

amithmath, 2021-10-10:
Actually, I am working on custom float, which I will modify real.c and real.h later, but before that I wanted to test whether new keyword added in c-common.c works or not. So I replaced keyword "float" with "fp_custom"  at line # 482  /gcc/c-family/c-common.c and build compiler and tested small program,  I am getting error: unknown type name 'fp_custom'

kito-cheng, 2021-10-11:
Maybe you could reference how ARM add new type for `__fp16` and `__bf16`:
https://github.com/riscv-collab/riscv-gcc/blob/riscv-gcc-10.2.0/gcc/config/arm/arm-builtins.c#L2639
https://github.com/riscv-collab/riscv-gcc/blob/riscv-gcc-10.2.0/gcc/config/arm/arm-builtins.c#L1836

kito-cheng, 2021-10-11:
I mean reference and implement that in RISC-V back-end.

amithmath, 2021-11-04:
Hello Jim/others: I am stuck at custom float encode and decode, I humbly request your assistance.

I am trying to incorporate in custom floats in RISCV-32 elf, I am encoding and assigning to image at line 2985 in https://github.com/riscv-collab/riscv-gcc/blob/5964b5cd72721186ea2195a7be8d40cfe6554023/gcc/real.c
I am decoding in line 2989 from const long *buf and assigning values to lines 3031-3034

   // custom logic...
   // I removed case statements
      r->cl = rvc_normal;
      r->sign = sign;
      SET_REAL_EXP (r, exp); // I am assigning unbiased exp
      r->sig[SIGSZ-1] = image | SIG_MSB;
	  
In my view, encode_ieee_single function send IEEE-754 format float value to FPU part of RISCV-32 and decode_ieee_single function decode IEEE-754 float format from FPU part of RISCV-32
	  
	  
But I am getting some different values. In order to debug the the functions/compiler I assigned some random value to decode function like:

      r->cl = rvc_normal;
      r->sign = 1;
      SET_REAL_EXP (r, 12); 
      r->sig[SIGSZ-1] = 0xDADB62D5; 	

And built the compiler.	  
	  
In my view, regardless of custom float values. I should always get real equivalent above assigned value. But I am not getting this value either; it seems to me there are other dependency functions which calculate real value decoded from above function? 	  

I look forward to hearing from you soon.

Many Thanks.

kito-cheng, 2021-11-04:
I would suggest move the discussion to gcc mailing list to get wider attention and feedback, and I gonna to close this issue since it's already out of scope of this repo, generally we only discuss RISC-V specific stuffs here.

closed 2021-11-04 by kito-cheng