riscv-collab/riscv-gnu-toolchain #891: Installation error on linux

gibchikafa, 2021-04-29:
I am getting the following error when installing on Linux with `make linux` . It cannot create the hard link. Any help?

`ln -f /home/vagrant/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/nptl/pt-crti.o /home/vagrant/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/nptl/crti.o
ln: failed to create hard link '/home/vagrant/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/nptl/crti.o' => '/home/vagrant/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/nptl/pt-crti.o': Operation not permitted
Makefile:564: recipe for target '/home/vagrant/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/nptl/crti.o' failed`

cmuellner, 2021-04-29:
Cannot reproduce here.
How did you invoke `./configure`, i.e. which arguments did you use?

AFAIK hard link creation is only possible if the user has read and write access to the existing file.
So the following commands can help you analyzing the situation:

* `whoami`
* `ls -l /home/vagrant/riscv-gnu-toolchain/build-glibc-linux-rv64imafdc-lp64d/nptl/pt-crti.o`

Also, check if you have enough free disk space.

gibchikafa, 2021-04-29:
Thanks for replying. It worked now. I dont know what the issue was but I just started with another fresh VM. But I suspect that it was because I cloned the repository into the VM and host shared folder.

TommyMurphyTM1234, 2022-09-27:
Seems to have been an issue with the original build environment.
Closing as invalid.

closed 2022-09-27 by TommyMurphyTM1234