riscv-collab/riscv-gnu-toolchain #892: Build toolchain under Linux for Windows using mingw-w64

ILoveSpeccy, 2021-04-30:
I'm trying to build toolchain on ubuntu 20.04 for Windows using mingw-w64:

```
./configure --prefix=/opt/riscvgcc --enable-multilib --with-host=i686-w64-mingw32 --without-system-zlib
make -j
```

and get the following error before libraries are built:
`/bin/bash: riscv64-unknown-elf-gcc: command not found`

/opt/riscvgcc is included in PATH.

Should I pay attention to something?

ILoveSpeccy, 2021-04-30:
Thanks!
I built a Linux version of the toolchain and added its bin folder into PATH. Now it works! But I don't know if it's the right way. 

p.S.
I have already tried to build the toolchain in Windows using msys2. It didn't work either (forgot the error message). Compilation under Windows feels ten times slower (antivirus is disabled).

jim-wilson, 2021-04-30:
It is easier to do the build on a linux machine than on a Windows machine, but riscv-gnu-toolchain doesn't have direct support for this.  The SiFive toolchain sources do though.  You could try looking at sifive/freedom-tools.  This is larger and more complicated to build than riscv-gnu-toolchain though.  There is also crosstool-ng.  This has support for canadian cross builds, but you might need to write a config file for the canadian cross riscv64-elf build.
[http://crosstool-ng.github.io/](url)
Read the docs if interested in this.

ILoveSpeccy, 2021-04-30:
Many thanks for the support!

ILoveSpeccy, 2021-05-01:
I was able to build it using the msys2 under Windows!
There were errors in gdb compilation and I turned it off (--disable-gdb). I built a working toolchain for picorv32 with the following commands:

```
../configure --prefix=/build/riscvgcc --with-multilib-generator="rv32e-ilp32e--c;rv32em-ilp32e--c;rv32i-ilp32--c;rv32im-ilp32--c" --disable-gdb
make -j$(nproc)
```

Thank you again!

closed 2021-05-01 by ILoveSpeccy