riscv-collab/riscv-gnu-toolchain #893: libfortran error

amithmath, 2021-04-30:
I am trying to compile simple fortran hello world program I am getting following error:
fatal error: cannot read spec file 'libgfortran.spec': No such file or directory

I used this command to compile: path/where riscv32-elf bin installed/gfortran hello_word.f90

I am wondering, what could be the problem?

I did: 'find -iname libgfortran.spec' but I could not get any files....

I am using gcc version 9.2.0

cmuellner, 2021-04-30:
Fortran support is only available in the Linux toolchain.

```
/opt/riscv-linux$ find . -name libgfortran.spec
./sysroot/lib64/lp64d/libgfortran.spec
./sysroot/lib64/lp64/libgfortran.spec
./sysroot/lib32/ilp32/libgfortran.spec
./sysroot/lib32/ilp32d/libgfortran.spec
./riscv64-unknown-linux-gnu/lib64/lp64d/libgfortran.spec
./riscv64-unknown-linux-gnu/lib64/lp64/libgfortran.spec
./riscv64-unknown-linux-gnu/lib32/ilp32/libgfortran.spec
./riscv64-unknown-linux-gnu/lib32/ilp32d/libgfortran.spec
```

amithmath, 2021-04-30:
The above problem I am facing for custom rv32 elf compiler I am developing. However, if I build rv32 elf compiler as it I am can find libfortran.spec and .f90 codes are successfully compiling. When I use find -iname libfortran.spec, I am getting files at following location:

./riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv32-unknown-elf/libgfortran/libgfortran.spec
./riscv32-unknown-elf/lib/libgfortran.spec

I am wondering am I missing for my custom rv32 elf compiler ?

amithmath, 2021-04-30:
The above problem I am facing for custom rv32 elf compiler I am developing. However, if I build rv32 elf compiler as it is from risc-v-gnu-toolchain github repo I am can find libfortran.spec and .f90 codes are successfully compiling. When I use find -iname libfortran.spec, I am getting files at following location:

./riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv32-unknown-elf/libgfortran/libgfortran.spec
./riscv32-unknown-elf/lib/libgfortran.spec

I am wondering am I missing for my custom rv32 elf compiler ?

cmuellner, 2021-04-30:
The `Makefile.in` includes Fortran support for the linux toolchain:
  https://github.com/riscv/riscv-gnu-toolchain/blob/master/Makefile.in#L402

The `Makefile.in` does not include Fortran support for the ELF/newlib toochain:
  https://github.com/riscv/riscv-gnu-toolchain/blob/master/Makefile.in#L623

jim-wilson, 2021-04-30:
> The `Makefile.in` does not include Fortran support for the ELF/newlib toochain:

Christoph, obviously he modified the Makefile to build Fortran support for the newlib toolchain.  This was discussed in a previous issue.

jim-wilson, 2021-04-30:
> I am wondering am I missing for my custom rv32 elf compiler ?

We don't have your custom rv32 elf compiler, so we can't say for sure what might be wrong.  We can only give hints on how to debug the problem.  Does the libgfortran.spec file exist in the install tree for this compiler?  Try using "gcc --print-search-dirs".  This will print the list of dirs that gcc will search for programs and libraries.  The libgfortran.spec file should be found one of the dirs in the libraries list.  If the libgfortran.spec file exists, is that dir in the libraries list?  Also try adding the -v option when running gcc, as this will give info about what the driver is doing.  All of this stuff works with g++ and gfortran same as it does with the gcc driver.

amithmath, 2021-04-30:
No  libgfortran.spec is not exist in install tree of this compiler. 

Yes I have modified the  Makefile and I have attached the Makefile.in in risc-v-gnu-toolchain
[Makefile_in.txt](https://github.com/riscv/riscv-gnu-toolchain/files/6406920/Makefile_in.txt)

amithmath, 2021-05-01:
All I am doing is applying patches from line number#41 (NEWLIB_URL)  in the attached Makefile along with other files and trying to build custom version of Fortran compiler for rv32 elf. It works fine for C,C++. The problem for  fortran is, it is generating fortran compiler but it is not generating libgfortran.spec and/or libgfortran in the installed tree, I am unable to figure-out where build is failing. I am badly stuck and I humbly request your assistance. I request you to copy these attached files and try to build fortran compiler, then you can figure out where I am failing. Many Thanks.
    
[Makefile.txt](https://github.com/riscv/riscv-gnu-toolchain/files/6409039/Makefile.txt)
[newlib.patch.txt](https://github.com/riscv/riscv-gnu-toolchain/files/6409040/newlib.patch.txt)
[riscv-gcc.patch.txt](https://github.com/riscv/riscv-gnu-toolchain/files/6409041/riscv-gcc.patch.txt)
[spike.patch.txt](https://github.com/riscv/riscv-gnu-toolchain/files/6409042/spike.patch.txt)

amithmath, 2021-05-01:
Also attached  modified Makefile.in in riscv-gnu-toolchain
[Makefile_in.txt](https://github.com/riscv/riscv-gnu-toolchain/files/6409199/Makefile_in.txt)

jim-wilson, 2021-05-01:
You need to look for errors in the build log.  You should be doing something like
make -jX &> build.log
whene X is some reasonable number for parallel jobs, and then looking for the string "error:" in the build.log file.  If you have trouble with that, you can try a build with -j1 and then the error will be at the end, but it will take a lot longer.

Looking at your Makefile.txt, I see a reference to Centos6 SCL devtoolset-2 (gcc-4.8) which is ancient.  That compiler may not be new enough to correctly and completely compile GCC 10.  It may be good enough to compile the C/C++ support, but not good enough to compile the Fotran support.  I'd suggest devtoolset-8 (GCC 8) instead.
https://www.softwarecollections.org/en/scls/rhscl/devtoolset-8/

amithmath, 2021-05-02:
OK, after installing devtoolset-8, now I am getting error in riscv-gnu-toolchain/build-gcc-newlib-stage2/riscv32-unknown-elf-dramfs/libbacktrace/config. log like this:

../riscv32-unknown-elf-dramfs/bin/ld: ../riscv32-unknown-elf-dramfs/lib/libgloss.a(sys_exit.o): in function `_exit ': sys_exit.c:(.text+0x2c): undefined reference to `dramfs_exit'

../riscv32-unknown-elf-dramfs/bin/ld:../riscv32-unknown-elf-dramfs/lib/libgloss.a(dramfs_fdtable.o): in function `.L0 ': dramfs_fdtable.c:(.text+0x13c): undefined reference to `dramfs_fs'

The thing is, these errors won't occur when I generate c/c++

I am scratching my head why it is failing....

jim-wilson, 2021-05-02:
libbacktrace is a required library for libgfortran.  It isn't built if you only configure for C and C++.

The dramfs_exit and dramfs_fs references are curious.  Those aren't symbol in libgloss.  Do you have libgloss patches that add them.  Or maybe there is an extra library that provides them?  If so, then you need to set up gcc to link with that extra library so that you can compile libbacktrace.

amithmath, 2021-05-03:
I added libbacktrace, in riscv-gnuMakefile.in if test -f $</contrib/download_prerequisites && test "@NEED_GCC_EXTERNAL_LIBRARIES@" == "true"; then cd $< && ./contrib/download_prerequisites; fi at stamps/build-gcc-newlib-stage2, yes sys_exit and dramfs aren't in riscv libgloss. 

in riscv-gnu-toolchain/.gitmodules:

I am replacing 

[submodule "riscv-newlib"]
path = riscv-newlib
url =  https://github.com/bespoke-silicon-group/bsg_newlib_dramfs.git

They are coming from these patches: https://github.com/bespoke-silicon-group/bsg_newlib_dramfs/tree/dramfs/libgloss/dramfs and https://github.com/bespoke-silicon
group/bsg_newlib_dramfs/tree/dramfs/libgloss/dramfs/dramfs

I was wondering, how to link riscv-newlib/libgloss/dramfs to libbacktrace? 

What puzzles me is it does not throw errors when I generate c,c++....

jim-wilson, 2021-05-03:
> What puzzles me is it does not throw errors when I generate c,c++....

C and C++ are closely related languages.  Your newlib port was probably written for C, and accidentally also works for C++.  But fortran is a different language with different conventions.  If you didn't do the newlib port correctly it won't work for Fortran.  Since it is your newlib port you need to fix it.

amithmath, 2021-05-04:
I built gfortran compiler by enabling languages only c,c++ in both stages: stamps/build-gcc-newlib-stage1 &2 and then I enabled languages c,c++, fortran in both stages then I built gfortran compiler. But when I compile I am getting this error:

lib/libgfortran.a(compile_options.o): can't link soft-float modules with single-float modules

jim-wilson, 2021-05-04:
EIther you made a mistake with multilibbing somewhere, or you are specifying an option at compile time for a multilib that you didn't build.  It is usually safest to use -with-arch= and --with-abi= options when configuring to get exactly the arch/abi combination you want to compile for.  That way you don't need to deal with multilibs.

amithmath, 2021-05-04:
I used --disable-linux --with-arch=rv32imaf --with-abi=ilp32f  along with this, I am trying to use this option: --enable-multilib --disable-float I am not sure, whether this could eliminate the above error?

amithmath, 2021-05-05:
I found the problem, all these libfortran issues coming-up because if I change in riscv-gnutoochain/Makefile.in:
from
NEWLIB_TUPLE ?= $(call make_tuple,$(XLEN),elf)

to

NEWLIB_TUPLE ?= $(call make_tuple,$(XLEN),elf-dramfs)

I am curious why it is so....?

jim-wilson, 2021-05-05:
There could be a configure script somewhere matching riscv*-*-elf or even *-*-elf which won't work if you use elf-dramfs instead of elf.  Configure scripts generally should be written to use elf* as this kind of renaming is allowed, but it is possible that this mistake was made someplace and we haven't caught it yet.

Or maybe something else is wrong.  You have made a lot of changes.  I can really only explain how the unmodified sources work.

amithmath, 2021-05-08:
I am trying to call fortran functions in c, is there any linker flags I need to use?

For example, I am trying to call add.f90 (adding two real*4  in ADD_NUMBER function) in main.c. In order to call add_number_(), I am compiling

riscv32-unknown-elf-gfortran -c add.f90
../riscv32-unknown-elf-gcc -c main.c
../riscv32-unknown-elf-gcc main.o add.o -o test
./test
I am not getting any output? Am I missing something?

closed 2021-05-09 by amithmath