riscv-collab/riscv-gnu-toolchain #894: LD_LIBRARY_PATH

wq911, 2021-05-01:
I met a make error when I try to build a RV64GC tool chain.

jim-wilson, 2021-05-01:
If LD_LIBRARY_PATH starts or ends with a colon, or has two colons in a row, that will cause the build to fail, as an empty entry in the list is treated the same as dot ("."), i.e. the current directory.

Unsetting LD_LIBRARY_PATH should work though.  Maybe you have something in a .bashrc or other shell startup file that is causing LD_LIBRARY_PATH to be set again in every new shell?  That will break configure, if it is adding dot to LD_LIBRARY_PATH, as it uses a lot of shell scripts.

Note that adding dot to LD_LIBRARY_PATH is a potential security risk, because if you cd into a directory owned by someone else, and run commands, they could trick you into running one of their shared libraries which can then do anything to your account and files.  You should only put full path names in LD_LIBRARY_PATH.

kito-cheng, 2021-05-02:
Set `LD_LIBRARY_PATH` to empty string doesn't means unset, you need `unset LD_LIBRARY_PATH`, you could verify by following command on bash:
```shell
if [ -z ${LD_LIBRARY_PATH+x} ]; then echo "LD_LIBRARY_PATH is unset"; else echo "LD_LIBRARY_PATH is set to '$LD_LIBRARY_PATH'"; fi
```

More example:
```shell
$ export LD_LIBRARY_PATH= 
$ if [ -z ${LD_LIBRARY_PATH+x} ]; then echo "LD_LIBRARY_PATH is unset"; else echo "LD_LIBRARY_PATH is set to '$LD_LIBRARY_PATH'"; fi
LD_LIBRARY_PATH is set to ''
$ unset LD_LIBRARY_PATH
$ if [ -z ${LD_LIBRARY_PATH+x} ]; then echo "LD_LIBRARY_PATH is unset"; else echo "LD_LIBRARY_PATH is set to '$LD_LIBRARY_PATH'"; fi
LD_LIBRARY_PATH is unset
```
Reference 
[1] https://stackoverflow.com/questions/3601515/how-to-check-if-a-variable-is-set-in-bash

TommyMurphyTM1234, 2022-09-27:
Questions/confusion about `LD_LIBRARY_PATH` and not an issue with the RISC-V GCC toolchain itself.

closed 2022-09-27 by TommyMurphyTM1234