riscv-collab/riscv-gnu-toolchain #895: make linux error about 'rv32emaf' and 'rv32imaf'

czhaneva, 2021-05-02:
I got an emergency, I need to build two types of compiler.
The first one is a compiler with only 'F' extension but without 'D' extension. To achieve this, I tried following commands:
`

1. mkdir build && cd build
2. ../configure --prefix=/home/myname/riscv_tools/rv32imafc --with-arch=rv32imaf --with-abi=ilp32f
3. make linux -j60

`
Then I got an Error, the following is the Errors
`checking for egrep... /bin/grep -E                                             
mips nios2 powerpc riscv glibc does not yet support systems with the F but not D extensions       
Makefile:319: recipe for target 'stamps/build-glibc-linux-rv32imafc-ilp32f' failed                                                                              
make: *** [stamps/build-glibc-linux-rv32imafc-ilp32f] Error 1
make: *** 正在等待未完成的任务....                                              
yes                                                                             
checking for sysdeps preconfigure fragments... aarch64 alpha arc arm csky hppa i386 m68k microblaze checking for grep that handles long lines and -e... /bin/grep                                                                               
checking for egrep... /bin/grep -E                                           
mips nios2 powerpc riscv glibc does not yet support systems with the F but not D extensions                                                                     
Makefile:302: recipe for target 'stamps/build-glibc-linux-headers' failed 
make: *** [stamps/build-glibc-linux-headers] Error 1
`
I noticed that there `mips nios2 powerpc riscv glibc does not yet support systems with the F but not D extensions  `
So I am a little confused that whether my request is reasonable. If reasonable, I hope for someone can help me out.
---------------------------------------------------
The second one is a compiler with embedding ('E' extension). To achieve this, I tried following commands:
`

1. mkdir build && cd build
2. ../configure --prefix=/home/myname/riscv_tools/rv32emadc --with-arch=rv32emadc --with-abi=ilp32e
3. make linux -j60

`
But I failed at at configure-gcc in build-gcc-newlib-stage1, I google this error, and found this URL: https://groups.google.com/a/groups.riscv.org/g/sw-dev/c/yXbZJwnFJ_U/m/f1RU6s0MCQAJ
I have no idea what is the riscv-next branch, I didn't find it in my code. This seems to be an unofficial implementation as well at 2018. So is there any official implementation now? and what should I do.

kito-cheng, 2021-05-02:
Both configuration are unsupported for glibc and linux kernel, it's not impossible to support, but need lots of effort to porting and verify.

TommyMurphyTM1234, 2022-09-29:
No update/activity in over a year.
And, to reiterate what @kito-cheng says, the "normal" expected architectures for Linux etc. are `rv32gc/rv64gc` or supersets of extensions.
But others are free and welcome to create PRs (here or upstream) to add support to the necessary components for other architectures.

closed 2022-09-29 by TommyMurphyTM1234