riscv-collab/riscv-gnu-toolchain #896: GCC for RVV intrinsics produces redundent vsetvli

zinechant, 2021-05-05:
This is rvv branch on commit 28271f03bb538d926ad2889dc8ad1b0cb1b3b45c.

Original source code. This is essentially a XXHash32 on a 4-byte word:
```
    size_t vl = vsetvl_e8m1(bytes);
    vuint32m4_t vx = vlse32_v_u32m4((const uint32_t*)data, 1, vl);
    vuint32m4_t vh = vmul_vx_u32m4(vx, PRIME3, vl);
    vh = vadd_vx_u32m4(vh, PRIME5, vl);
    vh = vor_vv_u32m4(vsll_vx_u32m4(vx, 17, vl), vsrl_vx_u32m4(vx, 15, vl), vl);
    vh = vmul_vx_u32m4(vh, PRIME4, vl);
    vh = vxor_vv_u32m4(vh, vsrl_vx_u32m4(vh, 15, vl), vl);
    vh = vmul_vx_u32m4(vh, PRIME2, vl);
    vh = vxor_vv_u32m4(vh, vsrl_vx_u32m4(vh, 13, vl), vl);
    vh = vmul_vx_u32m4(vh, PRIME3, vl);
    vh = vxor_vv_u32m4(vh, vsrl_vx_u32m4(vh, 16, vl), vl);
    vh = vand_vx_u32m4(vh, HASH_MASK, vl);
    vh = vsrl_vx_u32m4(vh, 2, vl);
```

objdump:
```
   ...
   104f8:	00097657          	vsetvli	a2,s2,e8,m1,tu,mu
   104fc:	012677d7          	vsetvli	a5,a2,e32,m4,tu,mu
   10500:	0a5f6607          	vlse32.v	v12,(t5),t0
   10504:	012677d7          	vsetvli	a5,a2,e32,m4,tu,mu
   10508:	012677d7          	vsetvli	a5,a2,e32,m4,tu,mu
   1050c:	012677d7          	vsetvli	a5,a2,e32,m4,tu,mu
   10510:	96cdc257          	vsll.vx	v4,v12,s11
   10514:	012677d7          	vsetvli	a5,a2,e32,m4,tu,mu
   10518:	a2c84457          	vsrl.vx	v8,v12,a6
   1051c:	012677d7          	vsetvli	a5,a2,e32,m4,tu,mu
   10520:	2a440257          	vor.vv	v4,v4,v8
   10524:	012677d7          	vsetvli	a5,a2,e32,m4,tu,mu
   10528:	00812783          	lw	a5,8(sp)
   1052c:	9647e257          	vmul.vx	v4,v4,a5
   10530:	012677d7          	vsetvli	a5,a2,e32,m4,tu,mu
   10534:	a2484457          	vsrl.vx	v8,v4,a6
   10538:	012677d7          	vsetvli	a5,a2,e32,m4,tu,mu
   1053c:	2e440257          	vxor.vv	v4,v4,v8
   10540:	012677d7          	vsetvli	a5,a2,e32,m4,tu,mu
   10544:	00c12783          	lw	a5,12(sp)
   10548:	9647e257          	vmul.vx	v4,v4,a5
   1054c:	012677d7          	vsetvli	a5,a2,e32,m4,tu,mu
   10550:	00d00793          	li	a5,13
   10554:	a247c457          	vsrl.vx	v8,v4,a5
   10558:	012677d7          	vsetvli	a5,a2,e32,m4,tu,mu
   1055c:	2e440257          	vxor.vv	v4,v4,v8
   10560:	012677d7          	vsetvli	a5,a2,e32,m4,tu,mu
   10564:	01012783          	lw	a5,16(sp)
   10568:	9647e257          	vmul.vx	v4,v4,a5
   1056c:	012677d7          	vsetvli	a5,a2,e32,m4,tu,mu
   10570:	a24cc457          	vsrl.vx	v8,v4,s9
   10574:	012677d7          	vsetvli	a5,a2,e32,m4,tu,mu
   10578:	2e440457          	vxor.vv	v8,v4,v8
   1057c:	012677d7          	vsetvli	a5,a2,e32,m4,tu,mu
   10580:	000107b7          	lui	a5,0x10
   10584:	fff78793          	addi	a5,a5,-1 # ffff <main-0x75>
   10588:	2687c457          	vand.vx	v8,v8,a5
   1058c:	012677d7          	vsetvli	a5,a2,e32,m4,tu,mu
   10590:	00200793          	li	a5,2
   10594:	a287c457          	vsrl.vx	v8,v8,a5
   ...
```

These vsetvlis look the same to me as <a2> is never changed.
Tried both -O2 and -O3, but these vsetvlis persist.

Maybe it's mentioned someplace, but is there a wiki/Mailinglist/... tracking the progress of GCC RVV support?
I learned from compilation errors that fractional MUL instructions are not yet supported.
Thanks for the help!

Regards,
Chen

jim-wilson, 2021-05-05:
There is no one actively working on it, so there is no progress.  Current focus is on the LLVM vector support.

The intent is to optimize away redundant vsetvli instructions, but we are apparently missing some cases.

zinechant, 2021-05-06:
Thanks for the reply.
Any chance you could provide a pointer/repo/page for the llvm rvv efforts? Would love to know more.

Regards,
Chen

zinechant, 2021-05-06:
Seems they are already in upstream.
https://github.com/llvm/llvm-project/tree/d480f968ad8b56d3ee4a6b6df5532d485b0ad01e/llvm/test/CodeGen/RISCV/rvv

I am closing this issue.

closed 2021-05-06 by zinechant