riscv-collab/riscv-gnu-toolchain #899: GCC test unstable: g++.dg/tls/thread_local4[g].C

cmuellner, 2021-05-10:
I observed, that there are two GCC testcase, that seems to be quite unstable:
  g++.dg/tls/thread_local4.C
  g++.dg/tls/thread_local4g.C

This manifests in the following fails when run with qemu (examples):
```
  FAIL: g++.dg/tls/thread_local4.C  -std=gnu++17 execution test
  FAIL: g++.dg/tls/thread_local4g.C  -std=gnu++14 execution test
  FAIL: g++.dg/tls/thread_local4g.C  -std=gnu++17 execution test
```
I'm observing this for (rv32imac ilp32 medlow) as well as (rv64imafdc lp64d medlow).
GCC is mainline (a week old or so), I am building with riscv-gnu-toolchain with multilib.

When testing this on the command line with
```
  /opt/riscv-mainline/bin/riscv64-unknown-linux-gnu-g++ -march=rv32imac -mabi=ilp32 -mcmodel=medlow -std=gnu++17 thread_local4g.C -static -pthread
  ./a.out
```
then I cannot reproduce the issue (I had this running in an endless loop for one hour).

Does anyone have an idea what is causing that?

TommyMurphyTM1234, 2022-11-28:
Just wondering if there has been any update on or resolution of this issue or if it still happens sporadically and mysteriously?

cmuellner, 2022-11-28:
I have not seen these fails for at least half a year.
But I don't know if that is just my environment that changed.

TommyMurphyTM1234, 2022-11-29:
Can somebody list the exact steps needed to run this single test repeatedly, please? From building the toolchain (if necessary - and if a CI build cannot be used as-is), to setting up the test suite/simulator environment, to running the single test repeatedly.

Thanks

kito-cheng, 2022-11-29:
IIRC this case might fail very randomly on qemu user mode, but we never figure out it's atomic code gen issue or qemu implementation issue...

cmuellner, 2023-10-18:
I haven't seen this error since last time I commented.
Let's close as fixed.