riscv-collab/riscv-gnu-toolchain #901: Build fails with `riscv-gcc/libcpp/expr.c:802:18: error: format not a string literal and no format arguments [-Werror=format-security]`

AntonLydike, 2021-05-18:
I'm using NixOs 20.09 with the following shell.nix:
```
with import <nixpkgs> {};

stdenv.mkDerivation {
  name = "riscv-env";

  buildInputs = [
    stdenv 
    binutils 
    cmake 
    pkgconfig 
    curl 
    glibc 
    autoconf 
    automake 
    python3 
    libmpc 
    mpfr 
    gmp 
    gawk 
    bison 
    flex 
    texinfo 
    gperf 
    libtool 
    patchutils 
    bc 
    libdeflate
    expat
  ];
}
```

I've configured as per the README:

```
> ./configure --prefix=(pwd)/../toolchain-out
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for grep that handles long lines and -e... /nix/store/rxkghln0km9axapv94iz3pcsln6ricms-gnugrep-3.4/bin/grep
checking for fgrep... /nix/store/rxkghln0km9axapv94iz3pcsln6ricms-gnugrep-3.4/bin/grep -F
checking for grep that handles long lines and -e... (cached) /nix/store/rxkghln0km9axapv94iz3pcsln6ricms-gnugrep-3.4/bin/grep
checking for bash... /nix/store/9ywr69qi622lrmx5nn88gk8jpmihy0dz-bash-4.4-p23/bin/bash
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /nix/store/fb09p7qvry3sigcjiqsddjg5axs810rb-curl-7.74.0-bin/bin/curl
checking for wget... /run/current-system/sw/bin/wget
checking for ftp... no
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed
```

When I run `make`, I get this error:
```
checking for ld used by GCC... ld
checking if the linker (ld) is GNU ld... yes
checking for shared library run path origin... done
checking for iconv... (cached) no
configure: updating cache ./config.cache
configure: creating ./config.status
config.status: creating Makefile
config.status: creating config.h
config.status: executing depdir commands
mkdir -p -- .deps
make[2]: Entering directory '/home/anton/projects/riscos/toolchain/build-gcc-newlib-stage1/build-x86_64-pc-linux-gnu/libcpp'
g++  -I../../.././riscv-gcc/libcpp -I. -I../../.././riscv-gcc/libcpp/../include -I../../.././riscv-gcc/libcpp/include  -g -O2 -W -Wall -Wno-narrowing -Wwrite-strings -Wmissing-format-attribute -pedantic -Wno-long-long  -fno-exceptions -fno-rtti -I../../.././riscv-gcc/libcpp -I. -I../../.././riscv-gcc/libcpp/../include -I../../.././riscv-gcc/libcpp/include   -c -o charset.o -MT charset.o -MMD -MP -MF .deps/charset.Tpo ../../.././riscv-gcc/libcpp/charset.c
g++  -I../../.././riscv-gcc/libcpp -I. -I../../.././riscv-gcc/libcpp/../include -I../../.././riscv-gcc/libcpp/include  -g -O2 -W -Wall -Wno-narrowing -Wwrite-strings -Wmissing-format-attribute -pedantic -Wno-long-long  -fno-exceptions -fno-rtti -I../../.././riscv-gcc/libcpp -I. -I../../.././riscv-gcc/libcpp/../include -I../../.././riscv-gcc/libcpp/include   -c -o directives.o -MT directives.o -MMD -MP -MF .deps/directives.Tpo ../../.././riscv-gcc/libcpp/directives.c
g++  -I../../.././riscv-gcc/libcpp -I. -I../../.././riscv-gcc/libcpp/../include -I../../.././riscv-gcc/libcpp/include  -g -O2 -W -Wall -Wno-narrowing -Wwrite-strings -Wmissing-format-attribute -pedantic -Wno-long-long  -fno-exceptions -fno-rtti -I../../.././riscv-gcc/libcpp -I. -I../../.././riscv-gcc/libcpp/../include -I../../.././riscv-gcc/libcpp/include   -c -o directives-only.o -MT directives-only.o -MMD -MP -MF .deps/directives-only.Tpo ../../.././riscv-gcc/libcpp/directives-only.c
g++  -I../../.././riscv-gcc/libcpp -I. -I../../.././riscv-gcc/libcpp/../include -I../../.././riscv-gcc/libcpp/include  -g -O2 -W -Wall -Wno-narrowing -Wwrite-strings -Wmissing-format-attribute -pedantic -Wno-long-long  -fno-exceptions -fno-rtti -I../../.././riscv-gcc/libcpp -I. -I../../.././riscv-gcc/libcpp/../include -I../../.././riscv-gcc/libcpp/include   -c -o errors.o -MT errors.o -MMD -MP -MF .deps/errors.Tpo ../../.././riscv-gcc/libcpp/errors.c
g++  -I../../.././riscv-gcc/libcpp -I. -I../../.././riscv-gcc/libcpp/../include -I../../.././riscv-gcc/libcpp/include  -g -O2 -W -Wall -Wno-narrowing -Wwrite-strings -Wmissing-format-attribute -pedantic -Wno-long-long  -fno-exceptions -fno-rtti -I../../.././riscv-gcc/libcpp -I. -I../../.././riscv-gcc/libcpp/../include -I../../.././riscv-gcc/libcpp/include   -c -o expr.o -MT expr.o -MMD -MP -MF .deps/expr.Tpo ../../.././riscv-gcc/libcpp/expr.c
../../.././riscv-gcc/libcpp/expr.c: In function ‘unsigned int cpp_classify_number(cpp_reader*, const cpp_token*, const char**, location_t)’:
../../.././riscv-gcc/libcpp/expr.c:802:18: error: format not a string literal and no format arguments [-Werror=format-security]
  802 |        0, message);
      |                  ^
../../.././riscv-gcc/libcpp/expr.c:805:39: error: format not a string literal and no format arguments [-Werror=format-security]
  805 |           virtual_location, 0, message);
      |                                       ^
cc1plus: some warnings being treated as errors
make[2]: *** [Makefile:224: expr.o] Error 1
make[2]: Leaving directory '/home/anton/projects/riscos/toolchain/build-gcc-newlib-stage1/build-x86_64-pc-linux-gnu/libcpp'
make[1]: *** [Makefile:2866: all-build-libcpp] Error 2
make[1]: Leaving directory '/home/anton/projects/riscos/toolchain/build-gcc-newlib-stage1'
make: *** [Makefile:521: stamps/build-gcc-newlib-stage1] Error 2
```

jim-wilson, 2021-05-19:
It looks like your system compiler has enabled some optional warning options that FSF GCC does not enable by default, and made them hard errors.  This will break the build of various programs ... including FSF GCC.  The workaround is to turn that error into a warning by adding -Wno-error=format-security somewhere appropriate, like CFLAGS or CC or maybe modify the riscv-gcc Makefile to add in the option there.

This isn't a problem with riscv-gnu-toolchain itself.  It is a problem with how your system compiler interacts with FSF GCC sources.  You could try consulting some NixOS forum for info on how to compile FSF GCC sources.

AntonLydike, 2021-05-20:
Thanks for the help, I got it building.

It turns out, nixos adds extra compiler flags, just as you said: https://github.com/NixOS/nixpkgs/issues/18995

I got the toolchain building using these options:

```
let pkgs = import <nixpkgs> {};
in
with pkgs.stdenv;
with pkgs.stdenv.lib;
pkgs.mkShell {
  name = "riscv-shell";
  nativeBuildInputs = with pkgs.buildPackages; [ bash gnumake cmake curl git utillinux ];
  buildInputs = with pkgs; [     
    stdenv 
    binutils 
    cmake 
    pkgconfig 
    curl 
    autoconf 
    automake 
    python3 
    libmpc 
    mpfr 
    gmp 
    gawk 
    bison 
    flex 
    texinfo 
    gperf 
    libtool 
    patchutils 
    bc 
    libdeflate
    expat
    gnumake
    pkgs.gnumake

    utillinux
    bash
    ];

  hardeningDisable = [ "all" ];
}
```
Notably, glibc is missing, as this caused a problem during compilation "stdlib.h not found" (or similar). I also added `hardeningDisable = [ "all" ];` to disable hardening flags for gcc.

I configured suing `./configure --prefix=$(pwd)/toolchain-out --with-arch=rv32im --disable-linux --disable-gdb --disable-multilib` and got it to build.

closed 2021-05-20 by AntonLydike