riscv-collab/riscv-gnu-toolchain #902: fatal: clone of 'git://sourceware.org/git/newlib-cygwin.git' into submodule path '/

maz-s, 2021-05-18:
OS: Ubuntu 18 on VMware fusion.
at installation (newlib) part, after I use "make," I have faced this problem:
[image]

jim-wilson, 2021-05-19:
Either the sourceware.org site was temporaily down, or you have a firewall on your side that prevents connecting to the sourceware.org site.  Try the command again.  Try visiting sourceware.org in your browser.

maz-s, 2021-05-19:
Thanks a lot for your kindly reply. I can reach  sourceware.org, and I have checked the firewall, the firewall is inactive.
I have tried installing riscv-gnu-toolchain base on the "https://github.com/riscv/riscv-gnu-toolchain" at Installation (Newlib) part. I am not able to change the git protocol.

jim-wilson, 2021-05-19:
Try using "git submodule update --init --recursive" in the riscv-gnu-toolchain top level dir.  Maybe there is a problem with the makefile commands trying to do this for you.  Maybe they don't work in your environment.  They do work for me.

If it is a firewall problem, you can change the protocol used in the .gitconfig file.  I have for instance
[url "ssh://git@github.com"]
        insteadOf = https://github.com
        insteadOf = http://github.com
        insteadOf = git://github.com
You can do the same thing with sourceware.org, and you can make any of the four protocols the default one.

Another approach to change the protocol is to edit the .gitmodules file in the riscv-gnu-toolchain source dir, and chance the sourceware newlib line.

If you don't need to build a toolchain, you can always download a pre-built one.  There are pre-built toolchains on the SiFive web site on the software page.

closed 2022-09-27 by TommyMurphyTM1234