riscv-collab/riscv-gnu-toolchain #903: riscv32-unknown-linux-gnu-gcc build 64-bit linux kernel?

CY-Pan, 2021-05-19:
I want to build a 32-bit riscv linux kernel for research. I cloned the toolchain and make the tools with arch rv32g. Then I downloaded the linux kernel and cross compiled it. But when I disassembled `vmlinux` with riscv32-unknown-linux-gnu-objdump, I found some 64-bit instructions. I checked it with `file` command and it also shows 64-bits. I can't figure out why. Are there restrictions in the tools, or in the kernel source code?

toolchain configuration:
[image]

64-bit instruction in the disassemble code of vmlinux:
[image]

check with `file` command:
[image]

jim-wilson, 2021-05-19:
In Kconfig, ARCH_RV64I is the default.  You need to change that to ARCH_RV32I via menuconfig or whatever.  RISCV_ISA_C, RISCV_ISA_A, and FPU are enabled by default, so those should be OK.

CY-Pan, 2021-05-20:
Thank you! Now I know how to get my solutions!

CY-Pan, 2021-05-20:
> Yeah, this isn't a toolchain bug.
> 
> `--with-arch` specifies the _default_ `-march` setting for the compiler, but it doesn't prevent `-march` from being overridden to build RV64 programs. Presumably, the Linux kernel is configured to build a 64-bit kernel, and so it is overriding `-march`.
> 
> You should check the kernel build configuration and make sure it knows it's building a 32-bit kernel. (Sorry, I can't help with that.)

Thank you!

CY-Pan, 2021-05-20:
> In Kconfig, ARCH_RV64I is the default. You need to change that to ARCH_RV32I via menuconfig or whatever. RISCV_ISA_C, RISCV_ISA_A, and FPU are enabled by default, so those should be OK.

Thank you! Now I know how to solve the problem!

ajf58, 2021-12-29:
@CY-Pan can this issue be closed?

closed 2021-12-30 by CY-Pan