riscv-collab/riscv-gnu-toolchain #905: Difference in ELF with PIE vs NOPIE

vlsunil, 2021-05-23:
I am seeing a discrepancy in the ELF file generated by the toolchain with and without --enable-default-pie option. 

Here is the sample program.

```
#include <stdio.h>
#include <stdint.h>
struct test {
    uint32_t m1;
    uint32_t m2;
    char name[64];
    uint64_t m3;
    uint32_t m4;
    uint32_t m5;
    unsigned long  m6;
    unsigned long  m7;
    const uint32_t *m8;
};

static uint32_t generic_m8[128] = { 0 };

struct test vls = {
    .m1 = 0x9,
    .m2 = 0x8,
    .name = "SUNIL",
    .m3 = 0x7,
    .m4 = 0x6,
    .m5 = 0x5,
    .m8 = generic_m8
};

int main()
{
    return 0;
}

```

When I compiled this sample program with the compiler built without --enable-default-pie, the ELF generated has contents like below.

```
0001000 0009 0000 0008 0000 **5553 494e 004c** 0000
0001010 0000 0000 0000 0000 0000 0000 0000 0000
*
0001040 0000 0000 0000 0000 0007 0000 0000 0000
0001050 0006 0000 0005 0000 0000 0000 0000 0000
0001060 0000 0000 0000 0000 **20a8 0001** 0000 0000
```
The data at the offset 0x1068 has the address of generic_m8 as expected.
```
$ readelf -a nopie | grep generic_m8
    45: 00000000000120a8   512 OBJECT  LOCAL  DEFAULT   21 generic_m8
```
However, the compiler built with --enable-built-pie (Like the Debian bullseye) generates the ELF with the below data for the same program.
```
0001000 0000 0000 0000 0000 0009 0000 0008 0000
0001010 5553 494e 004c 0000 0000 0000 0000 0000
0001020 0000 0000 0000 0000 0000 0000 0000 0000
*
0001050 0007 0000 0000 0000 0006 0000 0005 0000
0001060 0000 0000 0000 0000 0000 0000 0000 0000
0001070 **0000 0000 0000 0000** ffff ffff ffff ffff
```
The data at the offset 0x1070 which should have the address of generic_m8 variable is zero here. This is causing issues to the program at runtime if I refer to vls.m8 . 

Can someone with toolchain expertise help me here? Let me know if you need any other details. Thanks!

kito-cheng, 2021-05-23:
Are you running program on bare-metal environment? (or in another word, not on Linux/*BSD),  if so you need a program load for resolve those relocation, PIE means you can load program to anywhere of memory, but the price is you need to fix up those address when you load into program.

vlsunil, 2021-05-23:
Thanks @kito-cheng .

Yes, it is not running on Linux. However, I didn't see this behavior x86 toolchain for the same sample program (with and without pie). Is this different in riscv?

jim-wilson, 2021-05-23:
Relocation processing can vary from target to target.  This is not something you can rely on.  If using pie, you must perform relocations when loading the object into memory.

Nelson1225, 2021-05-24:
I think the difference is how to solve the RELATIVE relocation.

I remember that most of the targets used to encode the symbol values into GOT entries when enabling pie.  So that when the dynamic linker (or elf loader) resolving the RELATIVE relocation, they will load the original symbol values from the got entries first, and then plus the load offset and restore the values back to the got entries.  A total of one load and one store are required to resolve one RELATIVE relocation.

But RISC-V encodes the symbol value to the addend of RELATIVE relocation directly, so it doesn’t matter whether or not the symbol value is written into got entry.  Our dynamic linker will plus the load offset to the addend of RELATIVE relocation, and then store the value to got entries.  Therefore, we only need one store when relocating one RELATIVE relocation, this should reduce the burdens of dynamic linker.

As I know, x86 uses the former method, so they should always encode the symbol values into got entries with or without pie.  I’m not sure if it’s still the same now, since I haven’t seen the details in a while.

vlsunil, 2021-05-24:
Thanks a lot @jim-wilson and @Nelson1225 . This is very helpful.

closed 2021-05-24 by vlsunil