riscv-collab/riscv-gnu-toolchain #906: Newlib multilib build error

yavuz650, 2021-05-25:
I'm trying to build multilib for RV32I and RV32IM, both with ilp32 ABI. I ran the following sequence of commands to build the toolchain,

```
git clone https://github.com/riscv/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --with-multilib-generator="rv32i-ilp32-rv32im"
sudo make
```
However, at some point during the build, I get the following error,

```
/usr/bin/env: 'python': No such file or directory
invalid option for --with-multilib-generator
make[1]: *** [Makefile:4330: configure-gcc] Error 1
make[1]: Leaving directory '/home/yavuz/Downloads/riscv-gnu-toolchain/build-gcc-newlib-stage1'
make: *** [Makefile:521: stamps/build-gcc-newlib-stage1] Error 2
```

I'm on Ubuntu 20.04, and I have installed all the prerequisites. I copied the multilib command from Example 3 shown in the readme file.

What am I doing wrong here?

yavuz650, 2021-05-25:
> It seems like no python found in your system, you might need to install python-is-python3[1]:
> 
> ```
> $ sudo apt-get install python-is-python3
> ```
> 
> [1] https://askubuntu.com/questions/1144446/python-installed-in-ubuntu-but-python-command-not-found

Yeah I noticed and fixed that, but still got the error. Then, I tried doing `make clean` before doing `sudo make`. Now it seems to be installing with no trouble. It's still in the process of building though, I'll let you know if something goes wrong again. Thank you for the tip!

TommyMurphyTM1234, 2022-09-27:
No update/activity in over a year and it looks like the build issue was resolved anyway.

closed 2022-09-27 by TommyMurphyTM1234