riscv-collab/riscv-gnu-toolchain #907: 'Make' error for "rv32gcb" (Not able to build toolchain with b extension)

vidyatc, 2021-05-28:
hello all,
I'm trying to build the newlib toolchain with 32 bit using "--with-arch=rv32gcb" and getting the following error

Error: cannot find default versions of the ISA extension `b'

don't we have bit manipulation(b) support in the sources, 
cloned the sources from "git clone https://github.com/riscv/riscv-gnu-toolchain" (master).

Thanks in advance.

kito-cheng, 2021-05-28:
No, we don't have `b` support in `master` branch, you need to switch branch of `gcc` and `binutils` to `riscv-bitmanip` manually.

vidyatc, 2021-06-03:
Thanks, it worked.
When will this riscv-bitmanip go into the mainstream (master)?

jim-wilson, 2021-06-03:
When it gets ratified.  It is still a draft extension, and is still changing.  The zba/zbb/zbc/zbs subsets are supposed to go up for public review reall soon now, for a 45 day open review, followed by the voting for ratification.  But the rest of it, there is no way to know when it will be frozen and go through the ratification process.

closed 2021-06-08 by vidyatc