riscv-collab/riscv-gnu-toolchain #908: Compilation terminated error--reg

sreersr, 2021-05-28:
Hi,
I am trying to install "riscv-gnu-toolchain" in my ubuntu, but after compiling with "./configure --prefix=/opt/riscv", while I am trying to make it by using a "make linux" I am always getting error as attached here. Please help me to resolve this. Thanks in advance.

[image]

cmuellner, 2021-05-28:
Try installing `linux-libc-dev`.
Also, such trivial errors can be resolved by googling for the error message.

jim-wilson, 2021-05-28:
It is the RISC-V linux/errno.h that is missing.  Installing linux-libc-dev on the build machine won't fix that.

The RISC-V linux/errno.h should come from riscv-gnu-toolchain/linux-headers.  There should be a include/linux/errno.h file in there.  It should get installed when the build-linux-headers rule is run.  Check the sysroot to see if it looks OK.  It should contain a copy of every header file that is present in the source linux-headers dir.  If the file is present in the source tree, you can try forcing that rule to run again by "rm stamps/build-linux-headers; make stamps/build-linux-headers" and then check for the file in the sysroot again.

Note that if you have previously built a toolchain a while ago, and then updated, and try to rebuild and reinstall in the same dir that might not work unless you rm -rf the old install tree first.

TommyMurphyTM1234, 2022-09-29:
The `make linux` flow so I can only assume, as implied by @jim-wilson, that the issue here was with the user's build environment/host state which, most likely, needed a clean(er) setup. As such, and because there has been no update/activity on this issue for over a year, I'm closing this issue.

closed 2022-09-29 by TommyMurphyTM1234

lrettig, 2024-06-08:
> if you have previously built a toolchain a while ago, and then updated, and try to rebuild and reinstall in the same dir that might not work unless you rm -rf the old install tree first.

In case anyone else stumbles upon this: this happened to me because I had a failed build that was partially installed into a different prefix dir. Removing the old dir completely fixed the error.

TommyMurphyTM1234, 2024-06-09:
> In case anyone else stumbles upon this: this happened to me because I had a failed build that was partially installed into a different prefix dir. Removing the old dir completely fixed the error.

That's not the same as the case that @jim-wilson mentioned - **emphasis** is mine:

> if you have previously built a toolchain a while ago, and then updated, and try to rebuild and reinstall **in the same dir** that might not work unless you rm -rf the old install tree first.

In your case:

> In case anyone else stumbles upon this: this happened to me because I had a failed build that was partially installed into **a different prefix dir**. Removing the old dir completely fixed the error.

The contents of a previous different prefix dir should not impact the build into a different prefix dir. However what's left over in the build (not prefix) directories might. In any case it may make sense to do `make distclean` and also remove the current `prefix` dir if it exists from a previous build.