riscv-collab/riscv-gnu-toolchain #909: Git clone is stuck when not used with --recursive

tromedo, 2021-05-28:
Hi,

I am trying to build the toolchain and the git clone command is stuck when I don't get all the submodules by using recursive command. Worked when I cloned it with the --recursive flag. Just posting here because I thought it might be useful. Attaching the error message. 

flock `git rev-parse --git-dir`/config git submodule init /media/sriram/Data/experimental/riscv-gnu-toolchain/riscv-gcc/ && \
flock `git rev-parse --git-dir`/config git submodule update /media/sriram/Data/experimental/riscv-gnu-toolchain/riscv-gcc/
Submodule 'riscv-gcc' (https://github.com/riscv/riscv-gcc.git) registered for path 'riscv-gcc'
Cloning into '/media/sriram/Data/experimental/riscv-gnu-toolchain/riscv-gcc'...

After this point, it didn't progress.

lazyparser, 2021-05-29:
Hi,

Usually if we clone the repo using  `git clone --recursive` then we may stuck in QEMU submodule. for the riscv-gcc repo, if you see it is stuck, it is just downloading a huge amount of data (~1GB).

tromedo, 2021-05-29:
@lazyparser No, it's working fine when I use git clone --recursive. The problem is when I try to do it as given in the Github Readme. Says it will fetch the modules when needed. I used git clone without --recursive, then configured the options, executed the make command. That's where it tried to download the submodules and it got stuck forever. Then I used --recursive to manually get the packages and it worked. Downloaded around 6 GB. 

I hope I am making sense and not making a noob mistake :-D

LiuJiLan, 2021-07-17:
I ran into the same problem. 
I don't know if it is because the related statements in the README are misleading or because of network problems.
In my case, if I follow the suggestions in the README to clone and then build, or `git submodule update --init --recursive` after clone, I will encounter errors.
In short, for me, `git clone --recursive` can solve the problem.

Chris-Kimmel, 2021-08-14:
I had the same problem during [Newlib installation](https://github.com/riscv/riscv-gnu-toolchain#installation-newlib). After waiting a while, I aborted with Ctrl-C.

```
christopherlkimmel@penguin:~/projects/xv6/riscv-gnu-toolchain$ ./configure --prefix=/opt/riscv
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for grep that handles long lines and -e... /bin/grep
checking for fgrep... /bin/grep -F
checking for grep that handles long lines and -e... (cached) /bin/grep
checking for bash... /bin/bash
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... no
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed
christopherlkimmel@penguin:~/projects/xv6/riscv-gnu-toolchain$ make
cd /home/christopherlkimmel/projects/xv6/riscv-gnu-toolchain && \
flock `git rev-parse --git-dir`/config git submodule init /home/christopherlkimmel/projects/xv6/riscv-gnu-toolchain/riscv-gcc/ && \
flock `git rev-parse --git-dir`/config git submodule update /home/christopherlkimmel/projects/xv6/riscv-gnu-toolchain/riscv-gcc/
Submodule 'riscv-gcc' (https://github.com/riscv/riscv-gcc.git) registered for path 'riscv-gcc'
Cloning into '/home/christopherlkimmel/projects/xv6/riscv-gnu-toolchain/riscv-gcc'...

^Cmake: *** Deleting file '/home/christopherlkimmel/projects/xv6/riscv-gnu-toolchain/riscv-gcc/.git'
make: *** [Makefile:247: /home/christopherlkimmel/projects/xv6/riscv-gnu-toolchain/riscv-gcc/.git] Interrupt
```

I tried deleting the project directory and re-downloading it with `--recursive`:
```
git clone --recursive https://github.com/riscv/riscv-gnu-toolchain
```
This took an eternity, but there were progress bars. I then repeated the [Newlib installation](https://github.com/riscv/riscv-gnu-toolchain#installation-newlib). This time it seemed to be working, but I had to abort again because I ran out of hard-drive space.

This is all consistent with @lazyparser 's explanation: the installation is just very slow.