riscv-collab/riscv-gnu-toolchain #910: riscv32-unknown-elf-g++ -g hello.cpp -o hello

vinalin, 2021-05-29:
hello.cpp:
#include <iostream>
using namespace std;
int main(){
	cout<<"hello world!";
}

when i want to do the command riscv32-unknown-elf-g++ -g hello.cpp -o hello on ubuntu, it shows fatal error: iostream: No such file or directory
    2 | #include <iostream>

how can i fix it?

jim-wilson, 2021-05-29:
I would guess that there was a build failure during the stage2 gcc build.  The stage1 build for gcc builds the gcc/g++ binaries but not the C++ library.  The stage2 build for gcc builds the C++ library.  Check your build log for an error you might have missed, or try running make in the build dir again.  If the build didn't finish make will rebuild what failed the first time.

zhangping00, 2021-07-19:
[image]
May I ask why？

jim-wilson, 2021-07-19:
This doesn't appear to have any relationship to the previous question in this thread.  Please open new issues for new questions.

Also, there is no actual question here.  Just some gdb output.  You didn't mention what the target is (qemu?), and you didn't mention what the program being debugging is, or how it was compiled.  There is little we can do when we have no idea what exactly you are doing.  In general, we need enough info to reproduce.

Anyways, it looks like either your program is broken, or it was compiled wrong.

TommyMurphyTM1234, 2022-09-27:
No update/activity for over a year and the issue seems to be with the user's project/code and not a RISC-V GCC toolchain issue.

closed 2022-09-27 by TommyMurphyTM1234