riscv-collab/riscv-gnu-toolchain #911: Mac make project error, linker command failed with exit code 1

chipssss, 2021-05-30:
Hi, I am trying to make this project in mac os bigsur（11.7）, but happen some error as follow:

command:
> ./configure --prefix=/opt/riscv
> make

Warning && Error Output:
```
ld: warning: ignoring file ../opcodes/.libs/libopcodes.a, building for macOS-x86_64 but attempting to link with file built for unknown-unsupported file format ( 0x21 0x3C 0x61 0x72 0x63 0x68 0x3E 0x0A 0x2F 0x20 0x20 0x20 0x20 0x20 0x20 0x20 )
ld: warning: ignoring file /Users/chips/Desktop/OS/riscv-gnu-toolchain/build-binutils-newlib/zlib/libz.a, building for macOS-x86_64 but attempting to link with file built for unknown-unsupported file format ( 0x21 0x3C 0x61 0x72 0x63 0x68 0x3E 0x0A 0x2F 0x20 0x20 0x20 0x20 0x20 0x20 0x20 )
ld: warning: ignoring file ../bfd/.libs/libbfd.a, building for macOS-x86_64 but attempting to link with file built for unknown-unsupported file format ( 0x21 0x3C 0x61 0x72 0x63 0x68 0x3E 0x0A 0x2F 0x20 0x20 0x20 0x20 0x20 0x20 0x20 )
ld: warning: ignoring file ../libiberty/libiberty.a, building for macOS-x86_64 but attempting to link with file built for unknown-unsupported file format ( 0x21 0x3C 0x61 0x72 0x63 0x68 0x3E 0x0A 0x2F 0x20 0x20 0x20 0x20 0x20 0x20 0x20 )
Undefined symbols for architecture x86_64:
  "__bfd_elf_obj_attrs_arg_type", referenced from:
      _obj_elf_vendor_attribute in obj-elf.o
  "__bfd_std_section", referenced from:
      _main in as.o
      _dwarf2_directive_loc in dwarf2dbg.o
      _dwarf2_finish in dwarf2dbg.o
      _check_eh_frame in ehopt.o
      _make_expr_symbol in expr.o
      _expr_build_dot in expr.o
      _current_location in expr.o
      ...
// ...
 "_xstrerror", referenced from:
      _input_file_open in input-file.o
      _input_file_give_next_buffer in input-file.o
      _input_file_get in input-file.o
      _listing_print in listing.o
      _s_nop in read.o
      _s_func in read.o
      _stabs_generate_asm_func in stabs.o
      ...
  "_xstrndup", referenced from:
      _obj_elf_vendor_attribute in obj-elf.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[5]: *** [as-new] Error 1
make[4]: *** [all-recursive] Error 1
make[3]: *** [all] Error 2
make[2]: *** [all-gas] Error 2
make[1]: *** [all] Error 2
make: *** [stamps/build-binutils-newlib] Error 2
```

Thanks in advance.

xuanyuandy, 2021-08-25:
I met the same problem,do anyone know the solution,thanks very much !

jim-wilson, 2021-10-11:
This is apparently an issue with MacOS ld not accepting GNU ar output.  Or maybe the GNU/clang ld not accepting the MacOS ar output?  To solve this, you need to use the MacOS toolchain always, or the GNU toolchain always.  I see similar bugs in the FSF bugzilla but no solution.  There aren't many GNU developers with Mac access though.
[https://sourceware.org/bugzilla/show_bug.cgi?id=26193](url)

Perhaps using homebrew will fix this?

If you can't figure out how to build the toolchain, you can always download pre-built binaries.  There are multiple sources for that, e.g. the SiFive web site on the Software page has pointers to a number of pre-built toolchains for MacOS, free and commercial.

TommyMurphyTM1234, 2022-09-27:
No update/activity in almost a year and the issue seems to be yet another one of his best to approach building the toolchain on macOS (which seems to be using homebrew rather than anything else).

closed 2022-09-27 by TommyMurphyTM1234