riscv-collab/riscv-gnu-toolchain #912: Suggestion: discuss the ./configure option "--with-cmodel=medany" on the front README page

rsnikhil, 2021-06-11:
It seems that when building a compiler to use in RV64, one follows the README instructions (e.g., for multilib), often discovers that it doesn't work (for many setups, including Rocket's 0x_8000_0000 setup), searches the web for the "medany" diagnosis, and then rebuilds the compiler all over again so the libs get built with medany.  I understand one can't discuss all the options on the README, but this one seems important and frequent enough to warrant a brief mention, at least.

wineee, 2022-03-27:
I support this view...

cmuellner, 2022-03-28:
Sounds like a good idea.
Feel free to create a PR.
Thanks!

TommyMurphyTM1234, 2023-10-16:
No updates in over a year, no PR and no evidence of this being a common problem for users so I am closing this issue.

closed 2023-10-16 by TommyMurphyTM1234