riscv-collab/riscv-gnu-toolchain #913: Unable to understand R_RISCV_PCREL_HI20

suhasskrishanmurthy, 2021-06-15:
Hi,
   I am trying to compile a BareMetal application in riscv-tests/benchmarks repo.
I am getting this error
(.text.init+0x144): **relocation truncated to fit:** **R_RISCV_PCREL_HI20** against symbol `**_end**' defined in **.bss** section
/tmp/ccCBAFp0.o: in function `barrier':
syscalls.c:(.text+0x320): relocation truncated to fit: **R_RISCV_PCREL_HI20** against `count.2882'
collect2: error: ld returned 1 exit status

I am unable to understand the error message. Can anyone please help me in understanding this issue and in fixing this issue. will be a big help.
any links to understand the issue more properly is helpful and solving it.
My toolchain is,
The following options are target specific:
  -mabi=                      		**lp64d**
  -malign-data=               		xlen
  -march=                     		**rv64imafd**
  -mbranch-cost=N             		3
  -mcmodel=                   		[default]
  -mdiv                       		[enabled]
  -mexplicit-relocs           		[disabled]
  -mfdiv                      		[enabled]
  -mplt                       		[enabled]
  -mpreferred-stack-boundary= 		0
  -mrelax                     		[enabled]
  -mriscv-attribute           		[enabled]
  -msave-restore              		[disabled]
  -msmall-data-limit=N        		0
  -mstrict-align              		[enabled]
  -mtune=PROCESSOR            		rocket

  Supported ABIs (for use with the -mabi= option):
    ilp32 ilp32d ilp32e ilp32f lp64 lp64d lp64f

  Known code models (for use with the -mcmodel= option):
    medany medlow

  Known data alignment choices (for use with the -malign-data= option):
    natural xlen

Thanks in advance.

Nelson1225, 2021-06-24:
First, you can find all relocations and their meaning in the psabi,
https://github.com/riscv/riscv-elf-psabi-doc/blob/master/riscv-elf.md#pc-relative-symbol-addresses

The truncated error means linker cannot relocate the relocation, and usually caused by the overflow issue.  For R_RISCV_PCREL_HI20/LO12, the truncated error tells users that the range between the symbol and pc are too far, so that the 32-bit pcrel addressing isn't enough to represent the it.

One of the reason that causes the PCREL truncated error is the linker script.  According to the above error, linker try to tell us that the `_end` symbol is defined in the .bss section, and your linker script may place the data far away from the text.  Or maybe your linker script defines a very large alignment, including the data segment alignment, it may also cause the pcrel range much bigger than we thought.

jim-wilson, 2021-06-24:
Palmer Dabbelt wrote a blog series a few years ago that describes how the toolchain works, and part 2 talks about relocations.  You might find this useful.
https://www.sifive.com/blog/all-aboard-part-2-relocations

TommyMurphyTM1234, 2022-09-27:
Questions about `R_RISCV_PCREL_HI20` - not an issue with the RISC-V GCC toolchain itself.