riscv-collab/riscv-gnu-toolchain #917: error compile

MahdiBehkar, 2021-06-19:
/home/mahdi/riscv_tools_PATH/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccPW2N0U.o: in function `tohost':
(.tohost+0x0): multiple definition of `tohost'; /tmp/cc1HTnAS.o:(.sbss+0x10): first defined here
/home/mahdi/riscv_tools_PATH/lib/gcc/riscv64-unknown-elf/10.2.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccPW2N0U.o: in function `fromhost':
(.tohost+0x40): multiple definition of `fromhost'; /tmp/cc1HTnAS.o:(.sbss+0x8): first defined here
collect2: error: ld returned 1 exit status
make[1]: *** [/home/mahdi/riscv-tools/riscv-tools/riscv-tests/build/../isa/Makefile:74: rv32ui-v-simple] Error 1
make: *** [Makefile:29: isa] Error 2

kito-cheng, 2021-06-19:
I would suggest you create issue on `riscv-tests`, it's common issue on GCC 10 for legacy code, see also: https://stackoverflow.com/questions/66512831/compile-old-c-code-on-linux-that-throws-errors-multiply-defined-with-modern

closed 2021-06-19 by kito-cheng

MahdiBehkar, 2021-06-19:
I was able to compile  by adding flag -fcommon at the end of the Makefile in riscvtest/isa.
Thanks @kito-cheng

jim-wilson, 2021-06-21:
riscv-tools has an almost three year old copy of riscv-tests.  riscv-tools won't be useful until someone starts maintaining it again.  Meanwhile, if you use an up to date copy of riscv-tests it will work.