riscv-collab/riscv-gnu-toolchain #919: how to run the GCC Test Suite on windows ?

duchao713, 2021-06-23:
Hi,

I've compiled out a gcc.exe for windows.

Is it possible to run the GCC test suite on Windows? 
Just like what we do in Linux - "make report / check"

Thanks.

jim-wilson, 2021-06-24:
In theory, yes, you can run make check and make report on Windows.  The testsuites require dejagnu, tcl, and expect.  Dejagnu is included with the toolchain, at least until we upstream all bits of it that we need.  tcl and expect are open source packages that you will have to obtain from elsewhere.  If using something like cygwin or msys2 they should already have tcl and expect packages that you can use.  You just have to install them.  While this works in theory, it isn't something we test.

duchao713, 2021-06-25:
Thanks, I will try.

Is #858 doing something regarding this ?

jim-wilson, 2021-06-25:
858 is just talking about continuous integration (CI) testing.  It isn't trying to fix windows builds or checks.  Just trying to add automatic windows testing.

TommyMurphyTM1234, 2022-09-27:
Question about running the GCC test suite on Windows. Not a RISC-V GCC toolchain issue.

closed 2022-09-27 by TommyMurphyTM1234