riscv-collab/riscv-gnu-toolchain #920: GDB: sifive_e info frame error

Virtual-Machine, 2021-06-26:
### Summary
Hoping someone can either tell me what I am doing wrong or confirm that there is an issue here. I am planning to recompile from source soon anyways so if I should try a specific branch or version, let me know.

When using gdb against QEMU virt I have no issues.

When using gdb against QEMU sifive_e, I get an error every time I try to call "info frame".

### Issue
The error is preventing my command script from working correctly even though the error appears to be superficial. There may be a workaround for this but I am currently unaware.

### Error
Could not fetch register "time"; remote failure reply 'E14'

### Example
(gdb) i fr
Stack level 0, frame at 0x80000400:
 pc = 0x204000cc in start (kernel/start.c:18); saved pc = 0x20400026
 called by frame at 0x80000400
 source language c.
 Arglist at 0x80000400, args:
 Locals at 0x80000400, Previous frame's sp is 0x80000400
Could not fetch register "time"; remote failure reply 'E14'

### Versions
riscv64-unknown-linux-gnu-gdb --version
GNU gdb (GDB) 10.1
Copyright (C) 2020 Free Software Foundation, Inc.

qemu-system-riscv64 --version
QEMU emulator version 6.0.0
Copyright (c) 2003-2021 Fabrice Bellard and the QEMU Project developers

jim-wilson, 2021-06-26:
I think this is a communication problem between qemu and gdb concerning which csrs exist.  There was a fix added to gdb to ignore unknown csrs reported by a client.  And a fix was added to qemu to correct the csr xml register list it was sending to gdb.  Either fix should solve the problem.  But neither fix is in riscv-gnu-toolchain yet.  I think the gdb fix is not in a release yet, but in the soon to be released gdb 11, and I don't follow qemu so I don't know if that fix is in any release.

I think these are the right patches.  Either one should work if these are the right ones.
[https://lists.gnu.org/archive/html/qemu-devel/2021-01/msg02071.html](url)
[https://sourceware.org/pipermail/gdb-patches/2020-November/173566.html](url)

Virtual-Machine, 2021-06-26:
Thanks for the quick reply.

I am not very familiar with this type of patch link so I possibly did something wrong, please forgive my ignorance. I was able to copy the code from the GDB link  (modifying gdb/riscv-tdep.c) into my local copy and recompiled. I am still experiencing the same error message however so I either missed something critical or this change did not resolve it.

I assume there must be a way to apply one of these patches without copy pasting the code within the forum discussions sans the git diff symbols. I would try the QEMU patch if I knew how to apply it without copy pasting the changes I see in the discussion. I can't seem to see a download link or associated branch or anything I might be able to use but there may be a convention I am unaware of here.

jim-wilson, 2021-06-26:
It is easier to extract the patches from the git trees, but you need to know which patch you need and the above mailing list links show that.  The gdb home page is sourceware.org/gdb which points at sourceware.org/git/binutils-gdb.git and the qemu home page is qemu.org which points at https://gitlab.com/qemu-project/qemu.git.  They are both multipart patches, so you need all parts.

If this still doesn't work, then you might want to ask on an FSF gdb mailing list.  There are no gdb experts here.

Virtual-Machine, 2021-06-27:
I tracked down the QEMU changes and they were already applied to my copy of QEMU so it looks like I have uncovered a new issue possibly or I am doing something silly. I will try the GDB mailing list in a few days if I still have not resolved on my own.

For posterity, I got the 32 bit toolchain installed and I am getting the following error for both virt and sifive_e when trying to use "info frame".
Could not fetch register "ustatus"; remote failure reply 'E14'.

To summarize, gdb "info frame" response based on QEMU machine.

virt64 = no error
virt32 = "ustatus" error
sifive_e64 = "time" error
sifive_e32 = "ustatus" error

TommyMurphyTM1234, 2022-09-27:
@Virtual-Machine - do you consider this to still be a live issue or can it be closed?

TommyMurphyTM1234, 2022-11-28:
@Virtual-Machine, can you provide precise instructions on how to reproduce this issue if it's still extant, please? Otherwise it will be closed as not reproducible. Thanks.

TommyMurphyTM1234, 2022-12-03:
Closing this at not reproducible. 
If a reproducible test case/set of instructions is identified, please post them and reopen the issue.

closed 2022-12-03 by TommyMurphyTM1234