riscv-collab/riscv-gnu-toolchain #921: merge-newlib-nano copy fail on cross build

oferShinaar, 2021-06-27:
on makefile.in we have the following:
```
stamps/merge-newlib-nano: stamps/build-newlib-nano stamps/build-newlib
# Copy nano library files into newlib install dir.
	set -e; \
        for ml in `$(INSTALL_DIR)/bin/$(NEWLIB_TUPLE)-gcc --print-multi-lib`; \
	do \
```
When cross-building for none-Linux (Windows for this example), the command `$(INSTALL_DIR)/bin/$(NEWLIB_TUPLE)-gcc --print-multi-lib` will not work since the tool is already in `.exe` and the host is Linux; therefore, the copy and rename will not work.

oferShinaar, 2021-07-01:
how about the following fix?:
```	
stamps/merge-newlib-nano: stamps/build-newlib-nano stamps/build-newlib
# Copy nano library files into newlib install dir.
	set -e; \
        for ml in `$(NEWLIB_CC_FOR_TARGET) --print-multi-lib`; \
	do \
	    mld=`echo $${ml} | sed -e 's/;.*$$//'`; \
	    cp $(builddir)/install-newlib-nano/$(NEWLIB_TUPLE)/lib/$${mld}/libc.a \
		$(INSTALL_DIR)/$(NEWLIB_TUPLE)/lib/$${mld}/libc_nano.a; \
	    cp $(builddir)/install-newlib-nano/$(NEWLIB_TUPLE)/lib/$${mld}/libg.a \
		$(INSTALL_DIR)/$(NEWLIB_TUPLE)/lib/$${mld}/libg_nano.a; \
	    cp $(builddir)/install-newlib-nano/$(NEWLIB_TUPLE)/lib/$${mld}/libgloss.a\
		$(INSTALL_DIR)/$(NEWLIB_TUPLE)/lib/$${mld}/libgloss_nano.a; \
	done
```

kito-cheng, 2021-07-02:
The fix looks reasonable to me, but I guess that also require few other patches or precondition like required pre-built toolchain must in path with same multi-lib setting?

oferShinaar, 2021-07-04:
True.
When cross-building for windows, the `$(PATH)` needs to include the linux build.
You need it anyhow, mulltilib or not, that is why this fix is very generic and goes along with the makefiles design in this repo

TommyMurphyTM1234, 2022-10-15:
@cmuellner - can you outline to me how to use the riscv-gnu-toolchain build process to cross compile the toolchain for Windows in order to see if this issue is still extant?

I presume that I at least need to tell the build process to use a Windows toolchain such as `i686-w64-mingw32-gcc` but I can't see how to do this even having read through the `Makefile` created by the `configure` process.

cmuellner, 2022-10-17:
Sorry, but I am the wrong person to ask this, as I have zero experience in using GNU tools on a Windows machine (I just believe to know that there are three ways to do that: Cygwin, MinGW, and WSL), and I don't even have a Windows machine.

TommyMurphyTM1234, 2022-11-14:
Closing this as the instructions needed to reproduce it have not been provided.
If it's still and issue then please provide same and reopen.

oferShinaar, 2023-01-11:
Hi @cmuellner , @TommyMurphyTM1234 
Sorry for the late response. The team had a spinoff to a new company, so our activity on the open source stopped.
Yes its a Canadian Cross build, and it can also happen on other hosts (not only windows)
the configuration, in this case, was `--host=x86_64-w64-mingw32`
so if you want to reproduce it, you can do it from root config and makefile or from the `build-newlib-nano`

 **build-newlib-nano config:**
--target=riscv32-unknown-elf --prefix=<< build folder >>/install-newlib-nano --enable-newlib-reent-small --disable-newlib-fvwrite-in-streamio --disable-newlib-fseek-optimization --disable-newlib-wide-orient --enable-newlib-nano-malloc --disable-newlib-unbuf-stream-opt --enable-lite-exit --enable-newlib-global-atexit --enable-newlib-nano-formatted-io --disable-newlib-supplied-syscalls --disable-nls CFLAGS_FOR_TARGET="-Os -ffunction-sections s   -mcmodel=medlow" CXXFLAGS_FOR_TARGET="-Os -ffunction-sections -fdata-sections   -mcmodel=medlow"

From root:
--with-arch=rv32imac --with-abi=ilp32 --without-system-zlib **--with-host=x86_64-w64-mingw32 --with-multilib-generator=<provide few permutations**> --prefix=<< install folder >>

Since we use `NEWLIB_CC_FOR_TARGET` in this makefile, my suggestion is to keep using it also for the `nano merge`, basically to have homogeneous usage of `NEWLIB_CC_FOR_TARGET` when we need to use `-gcc`

TommyMurphyTM1234, 2023-01-11:
I don't know if it helps at all but, as far as I recall, this doesn't happen when Canadian Cross building the toolchain using crosstool-NG. It's a few months since I tried that but I can test again if it's if any use?

oferShinaar, 2023-01-11:
We are not using crosstool-NG. Do we want to tell users only to use crosstool-NG if they wish to do Canadian croos?

TommyMurphyTM1234, 2023-01-11:
> **From root:** --with-arch=rv32imac --with-abi=ilp32 --without-system-zlib **--with-host=x86_64-w64-mingw32 --with-multilib-generator=<provide few permutations**> --prefix=<< install folder >>

Did you change anything else to get it building? I get this on Zorin OS Lite 16.2 (Ubuntu 20.04 derivative):
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain riscv-gnu-toolchain-windows
cd riscv-gnu-toolchain-windows
./configure --prefix=`pwd`/installed-tools --with-host=x86_64-w64-mingw32
...
checking for the correct version of gmp.h... no
configure: error: Building GCC requires GMP 4.2+, MPFR 3.1.0+ and MPC 0.8.0+.
Try the --with-gmp, --with-mpfr and/or --with-mpc options to specify
their locations.  Source code for these libraries can be found at
their respective hosting sites as well as at
https://gcc.gnu.org/pub/gcc/infrastructure/.  See also
http://gcc.gnu.org/install/prerequisites.html for additional info.  If
you obtained GMP, MPFR and/or MPC from a vendor distribution package,
make sure that you have installed both the libraries and the header
files.  They may be located in separate packages.
make: *** [Makefile:559: stamps/build-gcc-newlib-stage1] Error 1
```
Did you edit any `Makefile` in order to get it to build the dependencies too?
If not can you provide full instructions on how to reproduce the issue please?
Thanks.

TommyMurphyTM1234, 2023-01-11:
> We are not using crosstool-NG. Do we want to tell users only to use crosstool-NG tool in case they wish to do Canadian cross?

I'm not sure if that's appropriate yet - or as long as development/improvement of standard extensions tools support is done in this repo. Even if it's possible to get CT-NG to build from sources other than official upstream etc. 

I would have to revisit my experimentation with CT-NG for building the toolchains, in particular Canadian Cross (build on Linux, run on Windows, target RISC-V, again to see what the lay of the land is these days. In general, I found CT-NG very useful but not without issues with regard to RISC-V toolchains. I'll try to have a look again soon.

oferShinaar, 2023-01-30:
@TommyMurphyTM1234  I did change the makefile. It's in at the being of this thread.
I used the commonly used convention for "target", in this makefile.in
[image]

Not to use the "installed" since it can be Windows or different host which can not run under the "builder" host

MarkGoncharovAl, 2023-03-05:
@oferShinaar 
I can't find PR to solve this problem - could you send here a link, I have the same error.
I would be very happy if this change will apply as soon as it can be. 
Also, I can create PR.

mga-sc, 2023-05-03:
My merged commit solved this problem, didn't it? @oferShinaar 

Please, check again and close this issue

TommyMurphyTM1234, 2023-08-19:
Does anybody have the precise instructions for reproducing this issue or, hopefully, checking thatt the aforementioned PR addresses it? I can't seem to find the magic combination. I thought that it should be this but no joy...
```
# Using WSL2/Ubuntu 22.04.3
# Install riscv-gnu-toolchain prerequisites
# Install gcc-mingw-w64 and g++-mingw-w64

git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-921
cd issue-921
export NEWLIB_CC_FOR_TARGET=x86_64-w64-mingw32-gcc
./configure --prefix=`pwd`/installed-tools --with-arch=rv32imac --with-abi=ilp32 --without-system-zlib --with-host=x86_64-w64-mingw32 --with-multilib-generator="rv32i-ilp32--,rv32im-ilp32--,rv32ima-ilp32--"
make 2>&1 | tee build.log
...
checking for the correct version of gmp.h... no
configure: error: Building GCC requires GMP 4.2+, MPFR 3.1.0+ and MPC 0.8.0+.
Try the --with-gmp, --with-mpfr and/or --with-mpc options to specify
their locations.  Source code for these libraries can be found at
their respective hosting sites as well as at
https://gcc.gnu.org/pub/gcc/infrastructure/.  See also
http://gcc.gnu.org/install/prerequisites.html for additional info.  If
you obtained GMP, MPFR and/or MPC from a vendor distribution package,
make sure that you have installed both the libraries and the header
files.  They may be located in separate packages.
make: *** [Makefile:588: stamps/build-gcc-newlib-stage1] Error 1
```

TommyMurphyTM1234, 2023-10-15:
> Does anybody have the precise instructions for reproducing this issue or, hopefully, checking that the aforementioned PR #1207 addresses it? I can't seem to find the magic combination. I thought that it should be this but no joy...
> 
> ```
> # Using WSL2/Ubuntu 22.04.3
> # Install riscv-gnu-toolchain prerequisites
> # Install gcc-mingw-w64 and g++-mingw-w64
> 
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-921
> cd issue-921
> export NEWLIB_CC_FOR_TARGET=x86_64-w64-mingw32-gcc
> ./configure --prefix=`pwd`/installed-tools --with-arch=rv32imac --with-abi=ilp32 --without-system-zlib --with-host=x86_64-w64-mingw32 --with-multilib-generator="rv32i-ilp32--,rv32im-ilp32--,rv32ima-ilp32--"
> make 2>&1 | tee build.log
> ...
> checking for the correct version of gmp.h... no
> configure: error: Building GCC requires GMP 4.2+, MPFR 3.1.0+ and MPC 0.8.0+.
> Try the --with-gmp, --with-mpfr and/or --with-mpc options to specify
> their locations.  Source code for these libraries can be found at
> their respective hosting sites as well as at
> https://gcc.gnu.org/pub/gcc/infrastructure/.  See also
> http://gcc.gnu.org/install/prerequisites.html for additional info.  If
> you obtained GMP, MPFR and/or MPC from a vendor distribution package,
> make sure that you have installed both the libraries and the header
> files.  They may be located in separate packages.
> make: *** [Makefile:588: stamps/build-gcc-newlib-stage1] Error 1
> ```
> 
> @oferShinaar - maybe you can check if that PR addresses your original issue?

Can anybody answer my questions here please?

TommyMurphyTM1234, 2023-10-15:
Unfortunately I don't know how to reproduce this issue to see if this PR fixes it:

* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1207

so I am going to have to trust that this confirmation is adequate to close it as resolved:

* https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1207#issuecomment-1466091462

closed 2023-10-15 by TommyMurphyTM1234