riscv-collab/riscv-gnu-toolchain #923: make: *** [Makefile:258: stamps/build-binutils-linux] Error 127

Heyuanhong, 2021-06-29:
when i try to compile rv32,my error as follow:
`mkdir -p /home/heyuanhong/riscv/.test || \
        (echo "Sorry, you don't have permission to write to" \
         "'/home/heyuanhong/riscv', use --prefix to specify" \
         "another path, or use 'sudo make' if you *REALLY* want to" \
         "install into '/home/heyuanhong/riscv'" && exit 1)`
`rm -r /home/heyuanhong/riscv/.test
mkdir -p stamps/ && touch stamps/check-write-permission
rm -rf stamps/build-binutils-linux build-binutils-linux
mkdir build-binutils-linux
cd build-binutils-linux && CC_FOR_TARGET=riscv32-unknown-linux-gnu-gcc /home/heyuanhong/riscv-gnu-toolchain/riscv-binutils/configure \
        --target=riscv32-unknown-linux-gnu \
         \
        --prefix=/home/heyuanhong/riscv \
        --with-sysroot=/home/heyuanhong/riscv/sysroot \
        --disable-multilib \
         \
        --disable-werror \
        --disable-nls \
        --with-expat=yes  \
        --disable-gdb \
        --disable-sim \
        --disable-libdecnumber \
        --disable-readline
/bin/sh: 1: /home/heyuanhong/riscv-gnu-toolchain/riscv-binutils/configure: not found`
`make: *** [Makefile:258: stamps/build-binutils-linux] Error 127`
my command as follow:
`heyuanhong@HECS:~/riscv-gnu-toolchain$ ./configure --prefix=/home/heyuanhong/riscv --with-arch=rv32i --with-abi=ilp32d`
`heyuanhong@HECS:~/riscv-gnu-toolchain$ sudo make linux`
so what's problem with my project?

jim-wilson, 2021-06-30:
Did you update submodules?  It looks like you checked out riscv-gnu-toolchain but not the submodules.  You can cd into the riscv-gnu-toolchain source tree and do "git submodule update --init --recursive" and that will download a lot of stuff.

rishav-singh-0, 2022-06-23:
I am also facing similar kind of issue, and yes i've tried to update submodules by using `git submodule update --init --recursive`. Still seeing the following error:
```
rm -rf stamps/build-binutils-newlib build-binutils-newlib
mkdir build-binutils-newlib
cd build-binutils-newlib && CC_FOR_TARGET=riscv64-unknown-elf-gcc /home/rishav/project/chromite/riscv-gnu-toolchain/riscv-binutils/configure \
	--target=riscv64-unknown-elf \
	 \
	--prefix=/home/rishav/project/chromite/riscv/riscv-newlib \
	 \
	--disable-werror \
	--with-expat=yes  \
	--disable-gdb \
	--disable-sim \
	--disable-libdecnumber \
	--disable-readline \
	--with-isa-spec=2.2
/bin/sh: 1: /home/rishav/project/chromite/riscv-gnu-toolchain/riscv-binutils/configure: not found
make: *** [Makefile:509: stamps/build-binutils-newlib] Error 127
```

Iamcaibirds, 2022-09-06:
I have this probelm too, don't know how to solve it

TommyMurphyTM1234, 2022-09-27:
If this is still a live issue can somebody please provide full details - including details of the build host system and all commands used to fetch, configure and build, and a full build log. Thanks.

TommyMurphyTM1234, 2022-09-27:
>  --with-arch=rv32i --with-abi=ilp32d

This doesn't make sense.
This combination of arch and abi is not valid.
`ilp32d` cannot be used with `rv32i` because the architecture does not include the D extension providing the required double floating point registers.
(Maybe `configure` could/should catch this, but that's another issue).

TommyMurphyTM1234, 2022-09-30:
Can't reproduce this and, in fact, the README says that no `update/recursive` commands are needed as the Makefile/build scripts take care of that. All that's needed is clone, configure and make.

closed 2022-09-30 by TommyMurphyTM1234

JacoboJin, 2023-12-07:
Unfortunately, this issue also occurs when I try to install the riscv-gnu-toolchain on PYNQ-Z2,  so how exactly can I fix this problem?

_What is certain is that I am not experiencing file loss._

```bash
mkdir -p /opt/rv32im/.test || \
	(echo "Sorry, you don't have permission to write to" \
	 "'/opt/rv32im'. Please make sure that the location is " \
	 "writable or use --prefix to specify another destination.'" \
	 && exit 1)
rm -r /opt/rv32im/.test
mkdir -p stamps/ && touch stamps/check-write-permission
rm -rf stamps/build-binutils-linux build-binutils-linux
mkdir build-binutils-linux
cd build-binutils-linux && CC_FOR_TARGET=riscv32-unknown-linux-gnu-gcc /home/xilinx/riscv-gnu-toolchain/binutils/configure \
	--target=riscv32-unknown-linux-gnu \
	 \
	--prefix=/opt/rv32im \
	--with-sysroot=/opt/rv32im/sysroot \
	--disable-multilib \
	 \
	--disable-werror \
	--disable-nls \
	--with-expat=yes  \
	--disable-gdb \
	--disable-sim \
	--disable-libdecnumber \
	--disable-readline \
	--with-isa-spec=20191213
/bin/sh: 1: /home/xilinx/riscv-gnu-toolchain/binutils/configure: not found
make: *** [Makefile:334: stamps/build-binutils-linux] Error 127
```

JacoboJin, 2023-12-07:
[out.txt](https://github.com/riscv-collab/riscv-gnu-toolchain/files/13599249/out.txt)