riscv-collab/riscv-gnu-toolchain #924: 'Fatal error: duplicate add.'

Qyqzsm, 2021-06-30:
Hi, I got a problem when I add a custom instruction to the binutil. Initially, I added this 2 line #define MATCH_ZSM 0x200002b #define MASK_ZSM 0xfe00707f and DECLARE_INSN(zsm, MATCH_ZSM, MASK_ZSM) to the **riscv-opc.h** file in binutils directory. For **riscv-opc.c** files in the directory I added {"zsm", "I", "d,s,t", MATCH_ZSM, MASK_ZSM, match_opcode, 0 } somewhere in between in "const struct riscv_opcode riscv_opcodes[] = ". Then I execute make clean and make in order. After around 30 minutes I got this error message: 'Fatal error: duplicate add.' 
I am really confused about the problem, and could someone pls help me? Thank you in advance!

jim-wilson, 2021-06-30:
In general, you need to debug this.  Run the assembler under gdb, put a breakpoint in as_fatal.  Then look at what happened and figure out what is wrong.

However, I would guess that you put your new instruction in the middle of the add instructions.  The assembler requires that all instructions with the same opcode name be consecutive in the riscv-opc.c table.  If you put zsm in the middle of the add instructions that will fail.  it needs to come between two instructions with different opcode names.

Qyqzsm, 2021-07-01:
> In general, you need to debug this. Run the assembler under gdb, put a breakpoint in as_fatal. Then look at what happened and figure out what is wrong.
>
> However, I would guess that you put your new [...]

Thank you jim, I have fixed the problem with the help of your suggestion and everything works fine now. Appreciate your quick and accurate response!

closed 2021-07-01 by Qyqzsm