riscv-collab/riscv-gnu-toolchain #927: Unable to compile opensbi v0.9 @17e23b678dac2e4579e714705cec137b351f7e9b with newlib

rogerta, 2021-07-10:
Hi there,

riscv-gnu-toolchain@995778bd3315faddd709948190ca76fb10b43188 is unable to compile opensbi@17e23b678dac2e4579e714705cec137b351f7e9b (v0.9) when configured with newlib.

riscv-gnu-toolchain was configured as follows:
```
$ ./configure --prefix=/opt/riscv
$ make
```
See attached [config.log](https://github.com/riscv/riscv-gnu-toolchain/files/6795652/config.log) file.

Trying to build opensbi produces the following error:
```
$ export CROSS_COMPILE=/opt/riscv/bin/riscv64-unknown-elf-
$ make PLATFORM=generic
...
 MERGE     platform/generic/firmware/payloads/test.o
 CPP       platform/generic/firmware/payloads/test.elf.ld
 ELF       platform/generic/firmware/payloads/test.elf
/opt/riscv/lib/gcc/riscv64-unknown-elf/11.1.0/../../../../riscv64-unknown-elf/bin/ld: -pie not supported
collect2: error: ld returned 1 exit status
make: *** [Makefile:351: /home/rogerta/src/opensbi/build/platform/generic/firmware/payloads/test.elf] Error 1
```
It's possible to get opensbi to compile by making this change to it (essentially removing `-pie` from cflags):
```
diff --git a/firmware/objects.mk b/firmware/objects.mk
index 67318e0..43939d6 100644
--- a/firmware/objects.mk
+++ b/firmware/objects.mk
@@ -20,9 +20,7 @@ endif
 ifeq ($(FW_PIC),y)
 firmware-genflags-y += -DFW_PIC
 firmware-asflags-y  += -fpic
-firmware-cflags-y   += -fPIE -pie
+firmware-cflags-y   += -fPIE
 firmware-ldflags-y  +=  -Wl,--no-dynamic-linker
 endif
```
I would expect riscv/riscv-gnu-toolchain should be able to build riscv/opensbi without modification to either.  Thanks.

rogerta, 2021-07-10:
Note that if I build the Linux cross-compiler instead of newlib then opensbi links correctly without changes.  Maybe worth adding a little more documentation to the README to explain when to use one cross compiler vs the other?

jim-wilson, 2021-07-11:
riscvXX-elf does not support -pie because no one ever implemented the newlib support necessary to make it work.  For a boot loader maybe you don't need any newlib support, but the missing newlib support breaks toolchain testing which causes problems, so -pie is off until someone does the newlib support.

I don't know anything about compiling opensbi.

jrtc27, 2021-08-11:
This should have been worked around in OpenSBI by https://github.com/riscv/opensbi/commit/09ad811ec4661b99a67b87f28c93177367ae911e (in that the default changes to not use PIE for this case), and as of https://github.com/riscv/opensbi/commit/81eb7088b6ae1b3b14ed57ef428b1b9f74d468cf it's documented that a bare-metal GNU toolchain cannot be used with the FW_PIC option.

(It seems OpenSBI upstream had only been testing with Linux GNU toolchains as otherwise they would have noticed they broke bare-metal ones a long time ago)

TommyMurphyTM1234, 2022-09-29:
No update/activity in over a year and the issue seems to be understood, addressed in OpenSBI, and not an issue with the RISC-V GCC toolchain itself.

closed 2022-09-29 by TommyMurphyTM1234