riscv-collab/riscv-gnu-toolchain #928: How to edit COLLECT_LTO_WRAPPER?

WeiyanZh, 2021-07-14:
Hi experts:

My OS: Ubuntu 20.04 LTS
I built RV32 successfully and put it in the _/opt/RV32_. 
Then I built RV64 successfully and put it in the _/opt/RV64_.

riscv32-unknown-elf-gcc works fine.
Now I want to use riscv64-unknown-elf-gcc. If I ran 
$ riscv64-unknown-elf-gcc -o hello hello.c

It returns error: 
hello.c:1:9: **fatal error: stdio.h: No such file or directory**

When I check $ riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
**COLLECT_LTO_WRAPPER=/usr/lib/gcc/riscv64-unknown-elf/9.3.0/lto-wrapper**
Target: riscv64-unknown-elf
Configured with: ../configure --build=x86_64-linux-gnu --prefix=/usr --includedir='/usr/include' --mandir='/usr/share/man' --infodir='/usr/share/info' --sysconfdir=/etc --localstatedir=/var --disable-silent-rules --libdir='/usr/lib/x86_64-linux-gnu' --libexecdir='/usr/lib/x86_64-linux-gnu' --disable-maintainer-mode --disable-dependency-tracking --target=riscv64-unknown-elf --prefix=/usr --infodir=/usr/share/doc/gcc-riscv64-unknown-elf/info --mandir=/usr/share/man ...

**Q:  I think it is linker problem. Should I change COLLECT_LTO_WRAPPER to /opt/RV64/bin/../libexec/gcc/riscv64-unknown-elf/11.1.0/lto-wrapper? and how? I tried to add PATH=$PATH:/opt/RV64/bin/ to the ~/.profile file, but it doesn't work.**

kito-cheng, 2021-07-14:
No, it's not linker issue, the error is emitted during compilation stage, gcc complain stdio.h not found, it generally happened when some thing wrong during build toolchain, I don't have too much information here, so I suggest you just do a clean build your rv64 toolchain...remove install folder and build folder and build again.

yanz08, 2021-07-14:
I tried to remove install folder and build toolchain again for several times. But the COLLECT_LTO_WRAPPER is still wrong. I still get the same error.
Under  _/usr/lib/gcc/_ there are two folders: riscv64-unknown-elf and x86_64-linux-gnu
I don't know why _/usr/lib/gcc/riscv64-unknown-elf/9.3.0/_ exists.

kito-cheng, 2021-07-14:
Maybe check your environment by `env` command? maybe some environment variable are set accidentally?

yanz08, 2021-07-14:
I solved it by adding **_PATH=/opt/riscv64imafdc/bin:$PATH_** instead PATH=$PATH:/opt/riscv64imafdc/bin/ in .profile file.

https://superuser.com/questions/238987/how-does-unix-search-for-executable-files explains why.

yanz08, 2021-07-14:
Thank you all!

closed 2021-07-14 by yanz08

jim-wilson, 2021-07-19:
Some OS distros have their own riscv64-unknown-elf-gcc package.  This includes recent Debian and Ubuntu releases.  You probably have the system compiler package installed in /usr/bin,   But the one in Ubuntu 20.04 is broken, it includes binutils and gcc but not newlib so is not very useful.  Putting your dir early in PATH will fix this as you noticed.

jrainzzz, 2023-07-20:
Hello, I also have the same problem. How did you solve the problem

TommyMurphyTM1234, 2023-07-20:
> Hello, I also have the same problem. How did you solve the problem

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/928#issuecomment-879781740

> I solved it by adding **_PATH=/opt/riscv64imafdc/bin:$PATH_** instead PATH=$PATH:/opt/riscv64imafdc/bin/ in .profile file.
> 
> https://superuser.com/questions/238987/how-does-unix-search-for-executable-files explains why.

And @jim-wilson's comments may also be relevant.

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/928#issuecomment-882850990

jrainzzz, 2023-07-20:
I did the same operation, but still couldn't solve it

TommyMurphyTM1234, 2023-07-20:
> I did the same operation, but still couldn't solve it

Try building the toolchain from scratch and using a local installation instead of the OS bundled one maybe?
```
cd ~/Downloads
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --enable-multilib
make

export PATH=`pwd`/installed-tools/bin:$PATH
# now try using the local build/install of the tools
...
```