riscv-collab/riscv-gnu-toolchain #929: undefined reference to __errno

suhasskrishanmurthy, 2021-07-15:
HI,
  I am trying to compile whetstone program. Am getting the following error
  I am not able understand the error. Can anyone help in understanding the error.
/home/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libm.a(lib_a-w_exp.o): in function `.L1':
w_exp.c:(.text+0x70): undefined reference to `__errno'
/home/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libm.a(lib_a-w_exp.o): in function `.L15':
w_exp.c:(.text+0x88): undefined reference to `__errno'
/home/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libm.a(lib_a-w_log.o): in function `.L0 ':
w_log.c:(.text+0x40): undefined reference to `__errno'
/home/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libm.a(lib_a-w_log.o): in function `.L1':
w_log.c:(.text+0x68): undefined reference to `__errno'
/home/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libm.a(lib_a-w_pow.o): in function `.L6':
w_pow.c:(.text+0xbc): undefined reference to `__errno'
/home/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /home/riscv/toolchain/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/lib/libm.a(lib_a-w_pow.o):w_pow.c:(.text+0x128): more undefined references to `__errno' follow
collect2: error: ld returned 1 exit status

jim-wilson, 2021-07-19:
You didn't mention the link command.  There is probably something wrong with it.  I would guess that you didn't use gcc to link, you instead tried to construct a link command by hand, and got it wrong.

The error is pretty simple, __errno, aka the C symbol errno, is undefined.  This should come from libc.  But there are dependencies between libc and libm, and static libraries are only search once in order by default, so if you include libc before libm, and nothing before libm referenced errno, then you will get an undefined error for errno at the end.  The solution is to mention libc again, e.g. -lc -lm -lc, or maybe put them in the other order, e.g. -lm -lc, but that isn't guaranteed to work, as there can be circular dependencies between libraries and worst case you might need to mention one or both multiple times.  The linker has a --start-group ... --end-group option pair to try libaries multiple times.

The knowledge of how to do this correctly is encoded in gcc, which is why it is always best to use gcc to link if you can.

yuxuan-z19, 2022-07-17:
Switching to `.cpp` from `.c` works for me somehow (hmm

TommyMurphyTM1234, 2022-09-27:
No RISC-V GCC toolchain issue identified. Seems more likely to be an issue with how the user code is being compiled/linked.